publiclab/spectral-workbench #414: New calibrations aren't available in Upload A Spectrum

carlcrott, 2018-10-13:
## What happened just before the problem occurred
Saved calibrations are available in the /capture/ but not in /upload/ 

## Relevant URLs
https://spectralworkbench.org/upload?webrtc=false
https://spectralworkbench.org/capture/

## PublicLab.org username
carlcrott

## Browser, version, and operating system
Safari 10.1.2 // OSX 10.10.5

Just posting here bc it seems like its an easy fix being that the data populates correctly to /capture/ ... so perhaps the /upload/ feature just needs the same database query as the /capture/ page.

carlcrott, 2018-10-13:
Looking through the network tab on Safari:

https://spectralworkbench.org/capture/recent_calibrations.json?calibration_id=144759

Seem to be the URL which is being pulled for my calibration data on /capture/ but not on /upload/

jywarren, 2018-10-17:
Thanks for reporting this! I think we should be able to come up with a fix
by looking at the capture controller and comparing the database call in
there to the one for the call you found. Are you familiar with Rails at
all? Thanks again!

carlcrott, 2018-10-18:
I know a bit. But if this is something thats not a 5-minute fix for someone who really knows rails well ... Its probably beyond what I can manage.