publiclab/spectral-workbench #415: log in issue

wangongtsen, 2018-11-01:
## What happened just before the problem occurred
I entered my PublicLab.org username into the log in form at https://spectralworkbench.org and hit the Log in button. Then I get redirected to a page saying: "We're sorry, but something went wrong. You may have mistyped the address or the page may have moved." With the URL below. The return_to part doesn't seem to make any difference.

## Relevant URLs

https://spectralworkbench.org/session/new?authenticity_token=dy7V%2Bv2IWNMjCb2jJ1qk%2F%2F%2B2FfC07u3bwwDUuHmpdC4%3D&back_to=%2F&open_id=felix-publiclab&return_to=%2F

## PublicLab.org username

felix-publiclab

## Browser, version, and operating system

Firefox 61 or Chromium 70 under Ubuntu 18.04

wangongtsen, 2018-11-01:
icarito suggested starting a private browser window or starting from a clean private browser window. This doesn't fix it for me though.
Perhaps I should mention, that I created that PublicLab.org account today at 13:05 UTC+1 and was never able to log into Spectral Workbench.

SidharthBansal, 2020-04-27:
I hope it works now. Please feel free to open up an issue again if problem persists

closed 2020-04-27 by SidharthBansal