publiclab/spectral-workbench #436: Service becomes unresponsive

icarito, 2019-02-07:
It was reported that SW failed to respond this morning. There was a **critical** alert for Load average at 4:36AM , but no alert for HTTP. It could be the Rackspace backup system which we know uses a lot of RAM. If it is the case we should prioritize switch backup systems.

icarito, 2019-02-07:
Here's the graph and endpoint list for this morning's slowdown:
[image]

Here's a link to the graph https://oss.skylight.io/app/applications/Wnv2jzIUWl5x/1549529820/18m/endpoints

jywarren, 2019-02-07:
In general if we may need to be moving off of Rackspace, we should maybe
prioritize a separate backup system anyways. Thanks, Sebastian!

icarito, 2019-02-08:
Thank you! Just now got another critical alert for high load on spectralworkbench, found the DB a bit busy:
[image]

icarito, 2019-02-08:
[image]
I'll check if skylight agrees! Perhaps there is a query we can optimize.

icarito, 2019-02-08:
Could it be setscontroller? 
[image]

https://www.skylight.io/app/applications/Wnv2jzIUWl5x/recent/6h/endpoints/SetsController%23show?responseType=json

jywarren, 2019-05-15:
I believe we've addressed this in #457!

closed 2019-05-15 by jywarren