publiclab/spectral-workbench #439: Severe slowdown at setsController.show

icarito, 2019-02-27:
Reported to web@ working group email.

[image]

icarito, 2019-02-27:
[image]

Going to check the logs to see what could cause such slowness...

icarito, 2019-02-27:
Application logs show:
```
WARNING: making https request to https://publiclab.org/openid without verifying server certificate; no CA path was specified.
  Rendered layouts/application.html.erb (2789.9ms)
Completed 200 OK in 3701.2ms (Views: 2639.8ms | ActiveRecord: 687.4ms)
kv_to_seq warning: Line 1 does not contain a colon: "<!DOCTYPE html>\n<html lang=\"en\">\n  <head>\n    <meta charset=\"utf-8\">\n    <title>Public Lab: a DIY environmental science community</title>\n    <meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0\">\n    <meta name=\"description\" content=\"Public Lab is an open community which collaboratively develops accessible, open source, Do-It-Yourself technologies for investigating local environmental health and justice issues.\">\n    <meta name=\"author\" content=\"Public Lab contributors\">\n    <link href=\"//publiclab.org/feed.rss\" r@@@              
```
Maybe this happens when logging in and publiclab.org itself is slow?

Also I see a lot of these:
```
Started GET "/capture/recent_calibrations.json?calibration_id=undefined" for REDACTED at 2019-02-24 09:11:33 +0000
```

jywarren, 2019-05-15:
I believe we've addressed this in #457!

closed 2019-05-15 by jywarren