publiclab/spectral-workbench #445: Cell phone cameras defaulting to wrong camera

bronwen9, 2019-04-07:
A user reported that spectral workbench (on phone) was defaulting to the front camera, which renders the papercraft spec unusable. I checked on my own phone (1st gen pixel) and found the same issue, with no obvious way to select the rear camera (this seems to be a newish issue, I hadn't experienced this before). 

https://publiclab.org/questions/Pepin420/04-06-2019/please-help

gauravano, 2019-04-08:
@jywarren this requires your attention. Thanks!

jywarren, 2019-04-08:
Hi! Yes, this is a known issue and we are trying to address it through a standard getUserMedia implementation -- tracking here and it's slated for Summer of Code as well! Thank you! https://github.com/publiclab/spectral-workbench.js/issues/87

closed 2019-04-08 by jywarren

gauravano, 2019-04-12:
Great!