publiclab/spectral-workbench #448: Frontend for SWB authentication

SidharthBansal, 2019-04-21:
As finalised in https://github.com/publiclab/mapknitter/issues/385, @IshaGupta18 is implementing the front end as 
[image]
for MK. We can use the same partial files here at SWB as the code should be exactly the same. 
Thanks all

SidharthBansal, 2019-04-21:
We will come to this once https://github.com/publiclab/mapknitter/issues/385 is complete @IshaGupta18 
I hope this works for you. I think we might be able to create a first timer issue for this one. Do you agree with me?

SidharthBansal, 2019-04-21:
I figured out there are some changes in the bootstrap layouts in SWB and MK so this issue is not ideal for fto eventhough code is similar. So, it has to be done by someone.

IshaGupta18, 2019-04-30:
Yes @SidharthBansal once the entire login system is complete in MK (with all PRs merged), I would split it up in small FTOs!

SidharthBansal, 2019-04-30:
Isha, MK PR is completely separate from the SWB PR.
Thanks
Sidharth

IshaGupta18, 2019-05-04:
@SidharthBansal #455 is my PR for this, kindly take a look at it. Thanks a lot!

PR #455 (2019-05-09): Login Modal redesign
Fixes #448 

Here is the design for the new login modal. I have deliberately kept the username field so that we can discuss if it's a good idea to remove it or not:

[image]
closed 2019-05-09 by jywarren