publiclab/spectral-workbench #451: Weekly Community Check-in #20 - What You Have Learnt

HarshKhandeparkar, 2019-04-29:
Hello Everyone!!!

We all at Public Lab :balloon:  - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. The goal of our Community Check-In is for every community member to share something about their work from the past week and about their current week's goal dart. You are also welcome to share fun-fact :smile: , new ideas :bulb: , your learning goals :ballot_box_with_check:

If you're new here, welcome :balloon: and please comment a Hello message below, we love to work with you. If you're looking for new issues, please try some of our first-timers-only issues.
You can find them at https://code.publiclab.org/#r=all

We're SO EXCITED to have your help!

Is there anything, you would like to share with us from last week's work? What is your plan for this week?

If you have not planned yet, just leave a Hello! so that we know that you are in sync with us :arrows_clockwise: and doing well!

As always, if you're waiting for a review, or if you're stuck, you can request help here, leave a comment on your issue with @publiclab/reviewers mentioned, or ask the community in our Gitter chatroom :open_hands:

**Gitter**

Gitter is an active chatroom in our community and we'll be sending weekly reminders about check-ins there. Be sure to sign up there for these updates or just to join the conversation page.

## **This Week's Theme**: 

**What You Have Learnt **: We all have experienced open source from the day we started our journey with PublicLab. PublicLab is known for welcoming new contributors so there is no doubt that your open source journey has been very interesting and got to learn many new things, especially about open source.
Apart from learning to contribute to open source projects, I learnt many other things and concepts (mostly mathematical concepts) which I had never expected to learn. I had never thought that a simple open source journey would teach me so many other things which are not at all related to programming.

This week, you can share what you have learnt that you never expected to learn:
- Did you learn a mathematical concept or something similar? Share your experience with us.
- Learnt something not related to programming at all? We would be eager to know what it is.
- Were you trying to solve one issue and ended up learning a lot more than just the solution to it? What did you learn?
- Something else? Share with us.

You can find our list of previous check-ins [here](https://github.com/publiclab/plots2/issues/5627) :notebook: 

### Google Season of Docs: 
 
**Applications for mentor (Google Season of Docs) are now open --- check out this exciting new program here:** https://github.com/publiclab/plots2/issues/5550

**Community Survey**

@ebarry has compiled the results of the community survey, which was open from January 14 to March 21, into a comprehensive code community report. Please add your comments here to this thread for what we are inspired to do/change as a result of reading:[https://publiclab.org/notes/liz/04-15-2019/report-2019-software-contributors-survey]( https://publiclab.org/notes/liz/04-15-2019/report-2019-software-contributors-survey)

This is a really unique opportunity to contribute to open source from a greater scope, and a reminder of the greater meaning behind the work all of you have been doing!! 

cc
@mohitRJranjan @milaaraujo @kevinzluo @Rishabh570 @sashadev-sky @geekychaser @shubhscoder @sidntrivedi012 @SidharthBansal @cesswairimu @stefannibrasil @ViditChitkara @tech4GT @mridulnagpal @IshaGupta18 @CoderJolly @namangupta01 @sagarpreet-chadha @Souravirus @MayankKashyap @siaw23 @ryzokuken @icarito @steviepubliclab, @ebarry, @jywarren, @sagarpreet-chadha, @JonathanXu1, @uzorjchibuzor, @eli6, @rexagod, @divyabaid16, @dinaelhanan, @oorjitchowdhary, @wanzulfikri, @mohitRJranjan @publiclab/mentors @publiclab/image-sequencer-guides @publiclab/leaflet-environmental-layers-guides @Paarmita @publiclab/reviewers @Divy123 @bhavayAnand9 @Mridul97 @avsingh999 @romanrodriguez @Dhiraj240 @codeIriss @rishabhc32 @chirag-singhal @IgorWilbert @dhruvgupta167 @vrk99 @pdurbin @HarshKhandeparkar @aashna27 @Harshithpabbati @marieram @swiatek7 
@niravasher

Thanks and keep contributing! Contributing to Public Lab means contributing to the Environment.

People who did the check-ins-

1. Gaurav Sachdeva (@gauravano) https://github.com/publiclab/plots2/issues/4327
2. Sidharth Bansal (@SidharthBansal) https://github.com/publiclab/plots2/issues/4369
3. Cess (@cesswairimu) https://github.com/publiclab/plots2/issues/4476
4. Harman Jolly (@CoderJolly) https://github.com/publiclab/plots2/issues/4548
5. Rishabh Rawat (@Rishabh570) https://github.com/publiclab/plots2/issues/4607
6. Sasha Boginsky (@sashadev-sky) https://github.com/publiclab/plots2/issues/4676
7. Rishabh Kr Bothra (@geekychaser) https://github.com/publiclab/plots2/issues/4711
8. Shubham Sangamnerkar https://github.com/publiclab/plots2/issues/4757
9. Siddhant N Trivedi (@sidntrivedi012) https://github.com/publiclab/plots2/issues/4784
10. Isha Gupta (@IshaGupta18) https://github.com/publiclab/plots2/issues/4844
11. Gaurav Sachdeva (@gauravano) https://github.com/publiclab/plots2/issues/4879
12. Harshith Pabbati (@Harshithpabbati) https://github.com/publiclab/plots2/issues/4917
13. Gaurav Sachdeva (@gauravano) https://github.com/publiclab/plots2/issues/5034
14. Harshith Pabbati (@Harshithpabbati) https://github.com/publiclab/image-sequencer/issues/883
15. Harman Jolly (@CoderJolly) https://github.com/publiclab/mapknitter/issues/447
16. Divya Baid (@divyabaid16) https://github.com/publiclab/PublicLab.Editor/issues/317
17. Gautami Gupta (@gautamig54) https://github.com/publiclab/leaflet-environmental-layers/issues/164
18. Sasha Boginsky (@sashadev-sky) https://github.com/publiclab/Leaflet.DistortableImage/issues/215
19. Sagarpreet Chadha (@sagarpreet-chadha) https://github.com/publiclab/leaflet-blurred-location/issues/198
20. Harsh Khandeparkar (@HarshKhandeparkar) 

Have a great week everyone :balloon: :tada: :sparkler:

HarshKhandeparkar, 2019-04-29:
@marieram @swiatek7 @gauravano @publiclab/reviewers @publiclab/is-reviewers @publiclab/community-reps

@subhahu123 @An0u @Jaya738 @mgroovyank @shreyateeza @TANGL3SIT3S @nazeem1988 @karunagoyalk @Nirvikalpa108 @cassianoblonski @EngrEric @Anjalizi @GettyOrawo @coderbeetle @corbettbw @govindjeevan @gautamig54 @daddyjab @emestabillo @letnewKid @srujana121 @jazdao @ChinwenduSilvia @aggarwal19 @Jenniline @madelinejones @helenatxu @richa031 @fabsar @kuja24 @megha070 @Aarabhi2017 @Manvi07 @siddhishree @Julius26 @VinneyJ @sonali9696 @rarrunategu1 @coreytegeler @labseven @milomacphail @CleverFool77 @ananyaarun @ananya @divyabaid16 @publiclab-mimi @shubhi2000 @Priyak5 @vaarigupta @techchic @ananya @hc-barker @edwd42 @PritiShaw @themonster2015 @supriya-kotturu @AnthoniaOkafor @jtmims @scheleon @mahmodHammad @santushk @jillpena @stoic-plus
@vaarigupta @alonpeer @madeofhuman @ibnjunaid @HarshKhandeparkar @aashna27 @Harshithpabbati @niravasher @mmmelissa

HarshKhandeparkar, 2019-04-29:
@Priyak5 @vaarigupta @techchic @ananya @hc-barker @edwd42 @PritiShaw @themonster2015 @supriya-kotturu @AnthoniaOkafor @jtmims @scheleon @mahmodHammad @santushk @jillpena @stoic-plus
@vaarigupta @alonpeer @madeofhuman @ibnjunaid @HarshKhandeparkar @aashna27 @Harshithpabbati @niravasher @mmmelissa

SidharthBansal, 2019-04-29:
I have learnt quite a lot from Public Lab
1. RoR
2. Testing
3. Workflow of SDLC
4. Mentorship
5. Collaboration, Team Bonding
6. Etiquettes, ethics while talking to seniors and juniors. Tone of speech.
It has increased my confidence and also helped me to improve my English
skills. I am not native English speaker.
7. Work on 3 repositories simultaneously I learnt this in last two months.
It is simply awesome experience to have 4-5 servers running simultaneously.
Thanks to @jywarren.
List goes on and on.....
So, it has been a wonderful experience.
Thanks all.

gauravano, 2019-04-30:
Hi everyone!

Great check-in @HarshKhandeparkar :+1: 

Contributing to Public Lab has surely made me a better developer, I learned testing, Git, understanding of diff. frameworks, being a good team player, collaboration skills, mentorship and many more. Although there is still a long way to go, there's always something to learn from every issue/PR/discussion/project.

I think I am also a bit more organized now, improved time-management skills(but still a lot more improvement needed :P), a good communicator, my typing and email reading speed have increased :sweat_smile:, increase in confidence and most importantly a better person.  

------
I would be slow in responding to issues/PRs till May 16, due to my exams, so if your PR is ready for review then please mention @publiclab/plots2-reviewers and if it's ready for merging then mention @jywarren @SidharthBansal. 

Thanks!
Have a great week ahead :sparkles:

skilfullycurled, 2019-05-02:
Again, another great topic!  It strikes me as funny that I never thought I would learn how to contribute make a contribution, make a pull request, go through testing, revisions, etc.   I had just never considered it in the first place.  Plots2 is a place where I've mainly contributed through ideas but it's just never been on my list of things to do.

skilfullycurled, 2019-05-02:
Again, another great topic!  It strikes me as funny that I never thought I would learn how to make a contribution,  (pull request, testing, revisions, etc).   I had just never considered it in the first place.  Plots2 is a place where I've mainly contributed through ideas but it's just never been on my list of things to do.

I also learned that the developer community is like any other community at Public Lab and that was a really crucial finding for me because it meant that I also learned that there was a good place for me to fit in which has been harder in recent years for their own interesting and necessary reasons. 

I think that's partly due to the way people conceive of websites in general.  When I hold an actual newspaper (which sadly, I do not) the printers are journalists.  It's where (I assume) "the press" comes from.  At a news website, developers disappear.  They don't produce news, instead they "maintain".  So, I think previously I've had the erroneous sense that Public Lab developers make the thing Public Lab does possible, instead of a thing which is made possible by Public Lab.  We too work on tools, research, activism, and so on.  I hope that in the future, we can find ways to interface in a more front facing way with the main website.

steviepubliclab, 2019-05-02:
Hi all!!! 
Heads up for those who haven't heard yet, we're doing an OpenHour this coming Monday about Launching Summer of Code. It will be at 14:30UDT, right after GSoC slot announcements are made. Hope you can hop on for a meet and greet. Also, big thanks to all who have applied for slots, volunteered to mentor, or are in other ways supporting the open coding community of Public Lab. Let's chat!  :)   www.publiclab.org/openhour   or direct link to join here: https://zoom.us/j/233841847

cesswairimu, 2019-05-03:
Awesome theme @HarshKhandeparkar.

Wow, where do I start? I have learnt so much here.
- Breaking up issues to mini-issues with the checklists
- Effective and friendly reviewing
- How to ask for help when stuck
- Collaboration
- How to think about how a non-technical person will use the project and try and make his experience better.
- How to make a DIY spectrometer :smile: 
Just to mention a few. Nice weekend everyone :balloon:

sashadev-sky, 2019-05-04:
Hi! I am going to go for one learning point instead of attempting to enumerate all the things i've learned from PL:

I wanted to remove the ability to "click spam" the like button on notes in plots2 and @jywarren taught me about debouncing in JS by pointing me to this PR for a usage reference: 
https://github.com/publiclab/plots2/pull/4904/ (thank you!)

I chose this one because it has proved to be a really useful and versatile JS skill, and I ended up learning so much beyond the original issue about optimizing client / server interactions.

sashadev-sky, 2019-05-04:
Is anyone interested in opening the next check-in tomorrow?

Divy123, 2019-05-04:
Yes please, I want to!!

aashna27, 2019-05-04:
Yeahh, I would like to volunteer to do it tomorrow.

sashadev-sky, 2019-05-04:
@Divy123 @aashna27 you guys commented at the same time, but @Divy123's comment is coming up 1st for me! 

Is it ok with the both of you if @Divy123 does this week and @aashna27 will have the week after? 

Also, feel free to open it in any repo you want

Divy123, 2019-05-04:
Thanks @sashadev-sky. I will be opening it tomorrow if it's okay to @aashna27?

sagarpreet-chadha, 2019-05-04:
Hey everyone ,
The openhour on monday is at 12:00 midnight Indian Time , right ? Can anyone confirm this ? Thank you :smile: !

rexagod, 2019-05-04:
@sashadev-sky @Divy123 @aashna27 If it's not a problem, I actually communicated my interest in opening  the [next check-in](https://github.com/publiclab/leaflet-blurred-location/issues/198#issuecomment-487421445) a while back to @gauravano, so can I proceed with this?

Divy123, 2019-05-04:
Sure please go ahead!!

gauravano, 2019-05-04:
Yes @sagarpreet-chadha, it's at 12:00 am IST

sashadev-sky, 2019-05-04:
@rexagod sorry I didn't realize! No problem it's all you. @Divy123 @aashna27 you guys get next two weeks :)

aashna27, 2019-05-04:
Okayy! Cool 😀

AnthoniaOkafor, 2019-05-05:
Sorry for the late response.
Publiclab is the first FOSS I'm contributing to and I have learnt a lot. I've learnt how to use Git, collaboration, time management and most important, I've learnt how to ask for help when stuck.

gauravano, 2019-05-05:
New check-in open at https://github.com/publiclab/inline-markdown-editor/issues/72. Thanks everyone!

closed 2019-05-05 by gauravano