publiclab/spectral-workbench #460: Update ReadMe to include Windows installation instructions

cryptoclidus, 2019-06-05:
## What happened just before the problem occurred
I was installing prerequisites to run spectral-workbench.

In the README file, the recommended ruby version manager (rvm) does not work on Windows without an elaborate workaround using a Linux cmd line interface, such as Bash for Ubuntu on Windows).
I am going to try to use an alternative (uru) to see if that makes it easier.

## Relevant URLs
https://rvm.io/rvm/basics

## PublicLab.org username
jenjimah

## Browser, version, and operating system
Windows10

jywarren, 2019-06-06:
Hi!! For a good generic rails setup guide please check out the links in
this issue! Sorry this repo is not as well documented as our others!!

https://github.com/publiclab/plots2/issues/3840

cryptoclidus, 2019-06-09:
Thanks for the link! That's a lot better for now. I got a lot more headway but still ran into trouble because neither rvm or rbenv would install 2.1.2, so I might have to try an earlier version that works with the repository. For now though, I've got electron running and developing on an app for me is a whole lot easier and seems more contained than working on the sw website directly.  Here's the link if you want to check it out! https://github.com/cryptoclidus/spectral-workbook/blob/master/README.md