publiclab/spectral-workbench #472: Remove white-space for footer

Uzay-G, 2020-01-07:
Hi, this is a [first-timers-only issue](https://code.publiclab.org/#r=all). This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [help wanted](https://github.com/publiclab/plots2/labels/help-wanted) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.
### The Problem
Currently, the footer for the website looks like this:
[image]

Could you help us center it and remove the whitespace?
To do this, you are going to lower the width of the text of the footer and you are going to style it with CSS to center it!
### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- Edit this footer css:
https://github.com/publiclab/spectral-workbench/blob/183f30cf825690506497f0b596b3745cc4f1ecef/app/assets/stylesheets/style.css#L113-L116
To add the following properties:
```css
footer {
  padding-bottom:200px;
  color:#777;
  margin: auto;
  width: 90%;
}
```

Then, change the classes of these `<div>` tags from `span4` to `span3`:
https://github.com/publiclab/spectral-workbench/blob/183f30cf825690506497f0b596b3745cc4f1ecef/app/views/layouts/application.html.erb#L75-L91

At the end, the footer will look like this!
[image]

[See this page](https://code.publiclab.org/#r=all) for some help in taking your first steps!

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:

1. If you are familiar with the terminal or would like to learn it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

2. You can also [edit files directly in your browser](https://help.github.com/articles/editing-files-in-your-repository/) and open a pull request from there. 

- [ ] 🏁 **Done** Ask in comments for a review :)

### Please keep us updated

💬⏰ - We encourage contributors to be respectful to the community and provide an update **within a week** of claiming a first-timers-only issue. We're happy to keep it assigned to you as long as you need if you update us with a request for more time or help, but if we don't see any activity a week after you claim it we may reassign it to give someone else a chance. Thank you in advance! 

If this happens to you, don't sweat it! Grab another open issue.

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it. 

👥- **If someone seems stuck, offer them some help!** Otherwise, [take a look at some other issues you can help with](https://code.publiclab.org/#r=all). Thanks!

### 🤔❓ Questions?

Leave a comment below!

developer22-university, 2020-01-12:
ok, @Uzay-G

developer22-university, 2020-01-12:
i am doing this @Uzay-G  thanks

developer22-university, 2020-01-12:
@Uzay-G  I want to work on other issues as well please guide

Uzay-G, 2020-01-12:
No problem @developer22-university! There are no longer many unassigned fto issues for now but maybe I can find another issue you could work on? Do you know a bit of Rails?

developer22-university, 2020-01-12:
yes, please review my PR. @Uzay-G

Uzay-G, 2020-01-18:
This was solved at https://github.com/publiclab/spectral-workbench/pull/479.

closed 2020-01-18 by Uzay-G