publiclab/spectral-workbench #478: Redesign search bar ui

Uzay-G, 2020-01-10:
The website search bar is quite small and I think it would be nice to redesign it a bit. Especially since when a user clicks the search icon, the search bar quickly appears below the menu and is very inconspicuous. I think we should make it stand out a little more and will work on implementing this.

Here is how it looks like at this point in time:
[image]

What do you think @publiclab/community-reps?

blurry-x-face, 2020-02-05:
Hi @Uzay-G I would like to work on it!

NitinBhasneria, 2020-02-28:
@cesswairimu what do you think about it?

najmieh, 2021-12-15:
@NitinBhasneria Can you assign this to me?

TildaDares, 2021-12-16:
Hi @najmieh, there's an issue in #10570 that is open if you're interested in working on it. You can leave a comment and I'll assign it to you.