publiclab/spectral-workbench #482: Weekly community Check-In #49 - Tech conferences recommendations

cesswairimu, 2020-02-18:
We all at Public Lab 🎈 - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. We have a Community Check-In each week, where every community member can share something about their work from the past week and about their current week's goal 🎯 . You are also welcome to share fun-fact 😄 , new ideas 💡 , your learning goals ☑️.

We believe in collaborative efforts to support our community. We are running a learning platform which helps a newcomer to become master of tomorrow. 💯

## Tech conferences to attend :microphone: 
I have had a couple of folks reach out asking for tech conferences to attend. This week, lets please give recommendations to great open conferences that people can attend. Give a name and city/country where the conference is happening. This may be of great help to past and current Outreachy and RGSoC interns who have a stipend for conference travel and also other contributors looking out for conferences. Who knows? you might meet a fellow contributor in your favorite conference :smile: 

## 2020 Software Contributor Community Survey Open :tada: 
Survey ticket is here: #7406
Don't forget our contributor community survey is still open! 

Please give us your thoughts in this anonymous form -- we'll post the results at the end and we can all learn about ourselves! We'll keep it open until March 27, 2020 https://forms.gle/Z6J3Mrdszm9iFiuv9

## Call for SOC mentors 2020 is still open
Mentors do crucial work during SOC and the application to be a mentor this summer is still open. More info here https://github.com/publiclab/plots2/issues/7360

Thanks everyone for making PL awesome and have a great week ahead :balloon:

cesswairimu, 2020-02-18:
@Rishabh570 @sashadev-skyl @vaibhavmatta @shubhscoder @aSquare14 @sidntrivedi012 @stefannibrasil @ViditChitkara @tech4GT @mridulnagpal @CoderJolly @namangupta01 @sagarpreet-chadha @Souravirus @MayankKashyap @siaw23 @ryzokuken @icarito @steviepubliclab, @ebarry, @jywarren, @sagarpreet-chadha, @JonathanXu1 @rexagod, @divyabaid16, @dinaelhanan, @oorjitchowdhary @publiclab/mentors @publiclab/image-sequencer-guides @publiclab/leaflet-environmental-layers-guides @Paarmita @publiclab/reviewers @Divy123 @bhavayAnand9 @Mridul97 @avsingh999 @Dhiraj240 @codeIriss @IgorWilbert @pdurbin @HarshKhandeparkar @niravasher @publiclab/reviewers @publiclab/is-reviewers @publiclab/community-reps @divyabaid16 @publiclab/community-reps @subhahu123 @govindjeevan @gautamig54 @rarrunategu1 @CleverFool77 @milaaraujo @kevinzluo @shipcy @developer22-university @keshavsethi @emilyashley @smarty1palak @daemon1024

cesswairimu, 2020-02-18:
@rehasantiago, @urvashigupta7, @HashTalmiz, @vinitshahdeo, @nstjean, @dragonwarp, @z-arnott, @VladimirMikulic, @NitinBhasneria, @Tlazypanda, @JanviMahajan14, @sssash18, @mantinone, @DouglasLutz, @seabl, @dependabot-preview[bot], @jywarren, @crisner, @Uzay-G, @shaibrown5, @kostogls, @pcube99, @imajit, @nicoleiocana, @simonastefan, @giovannyg, @priyanka-choubey, @anthony-zhou, @hknowlton, @Bhanu-mbvg, @pythongiant

alaxalves, 2020-02-26:
Hey guys, I'm not aware of any conference around where I live :laughing:, but I'll keep an eye out if there's any. For now I'm working on solving some minor bugs in Spectral Workbench, trying to get everything preset for GSoC. I've already gotten @SidharthBansal to mentor me on the upgrade project. I'll come up with a sketch proposal, with the plans I have regarding Spectral project and keep solving minor bugs.

emilyashley, 2020-02-28:
Oooh good question! Some of my favorite conferences have been FOSS4G, SOTMUS (openstreetmap), NACIS (Cartography), and JS.geo! Hahah can you tell I love maps and technology? I've always wanted to go to StrangeLoop tho!

I've spoken at FOSS4G, JS.geo, and ClosureSync and had a really good time at all of them :)

emilyashley, 2020-02-28:
@nstjean oh cool, Collision used to be held in New Orleans so I've been a few times. It's great for the intersection of tech and business and product. Very start-up oriented 😁

cesswairimu, 2020-03-03:
PyconKe and RubyconfKe  are great ones, they happen in Nairobi, Kenya but I heard they might be postponed this year because of the ongoing scares.

cesswairimu, 2020-03-03:
Moved to https://github.com/publiclab/leaflet-blurred-location/issues/237 Thanks all

closed 2020-03-03 by cesswairimu