publiclab/spectral-workbench #497: Weekly community Check-In #54 - How are you doing? Check-in with the us

cesswairimu, 2020-04-13:
We all at Public Lab :balloon: - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. We have a Community Check-In each week, where every community member can share something about their work from the past week and about their current week's goal :dart:. You are also welcome to share fun-fact :smile:, new ideas :bulb:, your learning goals :ballot_box_with_check:.

We believe in collaborative efforts to support our community. We are running a learning platform which helps a newcomer to become master of tomorrow. :100:

### How are you doing? Check-in with the us

#### Next Check-in
If you would like to open the next weekly check-in please comment below.

Thanks everyone for making PL awesome and have a great week ahead :balloon:

SidharthBansal, 2020-04-27:
Duplicate

closed 2020-04-27 by SidharthBansal

cesswairimu, 2020-04-29:
I didn't realize I created two issues...thanks @SidharthBansal for this catch :+1: