publiclab/spectral-workbench #498: Weekly community Check-In #54 - How are you doing? Check-in with the us

cesswairimu, 2020-04-13:
We all at Public Lab :balloon: - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. We have a Community Check-In each week, where every community member can share something about their work from the past week and about their current week's goal :dart:. You are also welcome to share fun-fact :smile:, new ideas :bulb:, your learning goals :ballot_box_with_check:.

We believe in collaborative efforts to support our community. We are running a learning platform which helps a newcomer to become master of tomorrow. :100:

### How are you doing? Check-in with the us

#### Next Check-in
If you would like to open the next weekly check-in please comment below.

Thanks everyone for making PL awesome and have a great week ahead :balloon:

cesswairimu, 2020-04-13:
@keshavsethi, @jywarren, @AyanChoudhary, @imajit,@NitinBhasneria, @nstjean, @marientje, @rutger-t, @Janet33, @TroydonAnabolic, @priyanshbalyan, @Medlineme4, @mandysGit, @msanden, @AidenOliver, @Janetsomehow, @AntroMarina, @llabake, @khushi02, @Alexjake660, @LarrySul, @nicoleiocana, @chrisdrums1, @Angelvicks, @Akanksha1212, @beladinaelfitri, @rid1hady, @shreyaa-s, @goamoth, @SusannaDiV, @yusofsf, @nanlliu, @Tlazypanda, @keshav234156, @WaldenJosh, @B4Lee, @VisnuGanth, @jessibreen, @alaxalves, @kamal-kaur04, @sssash18, @chinmay021, @Srija616, @AnnaMinakova, @greenbrown, @taysa91, @TaishiRoy, @yagmurbaydogan, @Shulammite-Aso, @Jigyasa-Kumari, @moliver4, @justinelai, @Snouzy, @stevenalila, @Digiox, @yandu31, @milosv1, @jamescapps, @Nevenem, @chirag-wadhwa, @yogeshjdeshmukh, @omizrahi99, @cesswairimu, @theOpetunde, @sailahonkanen, @robin-natale, @avidler, @ccyccyccy

cesswairimu, 2020-04-13:
@moliver4, @justinelai, @Snouzy, @stevenalila, @Digiox, @yandu31, @milosv1, @jamescapps, @Nevenem, @chirag-wadhwa, @yogeshjdeshmukh, @omizrahi99, @cesswairimu, @theOpetunde, @sailahonkanen, @robin-natale, @avidler, @ccyccyccy

cesswairimu, 2020-04-13:
Hey al :wave:  I am doing fine. One thing I am doing to keep sane is, I stopped consuming too much news. I tried yoga last week, found out I don't have enough patience :laughing: ...will try again this week. I am also Netflixing.  
Great week ahead

Medlineme4, 2020-04-13:
Hello Everyone. I am doing great. I never knew staying at home could be this interesting.  I'm trying new things out. I do lots of exercise these days. And I am also happy I have the Public Lab Community to communicate with.

gauravano, 2020-04-13:
Hi everyone, 

I am doing great. 

Working out regularly from home, spend some time on the terrace (try to get a bit of sunlight daily), and then do office work. I have also started contributing to open source from yesterday. 

Wishing you all a happy week ahead 💯🥂

sashadev-sky, 2020-04-13:
hi everyone! 

To stay sane I have also been contributing to OS, and have found a lot more time to do so outside of my day job which I am happy about. 

Also I have taken up playing Animal Crossing; this quarantine activity was recommended to me by many people and lives up to its rep 😅

cesswairimu, 2020-04-13:
Hi @sashadev-sky and @gauravano nice to hear from you both its been a minute.. @sashadev-sky I will checkout animal crossing :smile:

Uzay-G, 2020-04-13:
Hey! I have been getting into competitive programming a lot during my quarantine and I have also been doing a few hackathons with some cool friends where we built a [github integration](https://reposi.0cdn.me). I also will have more time to hopefully contribute and build some nice stuff :smile:

alaxalves, 2020-04-13:
Last week I have been busy accomplishing some personal projects. Had not much time to contribute with the PL community, but I got to successfully open a PR at the Mapknitter-Exporter repo. This is the the link: publiclab/mapknitter-exporter#27. I intend to keep up with export-related contributions as well as the Rails upgrade going on in Spectral Workbench here --> https://github.com/publiclab/spectral-workbench/pull/495. Talked to @gr455 today and he agreed on collaborating with me on that.

Just finished the _Narcos_ series(great one btw) and started a comedy anime called _Seitokai no Yakuidomo_, which is pretty funny. Be safe guys.

**PS: Could someone kindly review https://github.com/publiclab/mapknitter-exporter/pull/27 ???**

sashadev-sky, 2020-04-13:
@cesswairimu you too!!! So awesome to so many familiar faces

gauravano, 2020-04-13:
Same here @cesswairimu. 

The project is looking good @Uzay-G. All the best for CP.

Uzay-G, 2020-04-13:
Thanks a bunch @gauravano!

SusannaDiV, 2020-04-14:
Hello, thanks for asking! I have online classes to keep me occupied, and I've started preparing myself for next year's SWERC with my colleagues. Living in Italy I haven't been outside once in the last 2 and a half month - I'm following 80's workout video tips to keep me active ahah!

B4Lee, 2020-04-15:
Hi Everyone,
I am doing great
Currently, I am busy working on e-commerce projects, but sometimes I play DotA 2 with my friends during this pandemic COVID 19 Quarantine.

I hope you all stay healthy and remember **#StayHome #SaveLives**

gr455, 2020-04-15:
Hi everyone!
Currently, I am helping @alaxalves with rails upgrade for spectral. Apart from that I have been learning django which is also a very cool web framework(also I am quite familiar with MVC through rails).
I play some video games in my spare time, just finished Hollow Knight(amazing game btw) and have kinda started reading Harry Potter :)

cesswairimu, 2020-04-18:
Thanks everyone for checking-in...Would anyone like to open next week's check-in? Thanks

goamoth, 2020-04-18:
Hola!
Good to hear from everyone, hope you are all doing great and keeping safe!
Am at home too and trying to take as much courses on Pluralsight as I can as this month all courses are free... Also been doing practice building react components using TDD with Jest. Love the whole test first approach, and I have a curious eye for JAMStack, still trying to understand how it works. Most importantly, am enjoying the new experience of contributing to open source. 
Glad to be part of this mailing list!

TroydonAnabolic, 2020-04-19:
Not managing much time to work on my projects now that im working full time and it takes 3 hours + of my time on travel. Then I have to smash some weights at home to maintain my physique. Hoping that my job goes to mainly remote work soon so I can use those extra 3 hrs to build fun Unity game and software app projects.

Angelvicks, 2020-04-23:
Hello everyone,  I am doing great now, was sick for a while. Currently increasing my skills in python and am learning java at the same time during this quarantine.

cesswairimu, 2020-04-28:
Hi all... I am out of themes and no one expressed interest of opening a new check-in...please check-in here for any concerns and questions this week. we shall open a new check-in next week. Have a great week ahead.

cesswairimu, 2020-04-28:
@Tlazypanda save the day :tada:  :tada:  Thanks
Moved to https://github.com/publiclab/plots2/issues/7844 
Thanks everyone

closed 2020-04-28 by cesswairimu