publiclab/spectral-workbench #520: Weekly Community Check-In #65 - GSoC First Eval Due this week

cesswairimu, 2020-06-30:
Hi everybody! 👏

We all at Public Lab 🎈 - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. We have a Community Check-In each week, where every community member can share something about their work from the past week and about their current week's goal 🎯 . You are also welcome to share fun-fact 😄 , new ideas 💡 , your learning goals ☑️.

We believe in collaborative efforts to support our community. We are running a learning platform which helps a newcomer to become master of tomorrow. 💯

## GSoC First Eval Due this week

**To all the GSoC interns:** This week we have GSoC First Evaluations please fill out the forms before the deadline. Ping the mentors if you have any questions on the evals.

**To all interns:**  I would like to commend you of all the great job and progress you are making on your projects.. we had a couple of releases last week :rocket: keep it up :tada:  :tada:... As usual please let us know what you are working on this week and if you have any blogs kindly drop the links.

**To all contributors:**  Please let us know how you are doing and if blocked on anything let us know so we can help. Thanks everyone taking their time to give reviews and participate in all the discussions going around the repos :tada: :tada: 

## Next Check-in
If you would like to open the next check-in..Leave a comment below. We are happy to help if its your first time

**Thanks everyone for making Public Lab awesome** 🎈
Have a great week and year ahead all ❤️

cesswairimu, 2020-06-30:
@JapneetSingh5, @Neilblaze, @markkulube, @vinuch, @Tlazypanda, @icarito, @dependabot-preview[bot], @jywarren, @keshavsethi, @praaatik, @Janitham97, @imanishbarnwal, @dburugupalli, @adityakpranav, @ioampatzidis, @esiebomaj, @Alexjake660, @cesswairimu, @urvashigupta7, @daemon1024, @jscastanos @keshavsethi, @jywarren, @AyanChoudhary, @imajit,@NitinBhasneria, @nstjean, @marientje, @rutger-t, @Janet33, @TroydonAnabolic, @priyanshbalyan, @Medlineme4, @mandysGit, @msanden, @AidenOliver, @Janetsomehow, @AntroMarina, @llabake, @khushi02, @Alexjake660, @LarrySul, @nicoleiocana, @chrisdrums1, @Angelvicks, @Akanksha1212, @beladinaelfitri, @rid1hady, @shreyaa-s, @goamoth, @SusannaDiV, @yusofsf, @nanlliu, @Tlazypanda, @keshav234156, @WaldenJosh, @B4Lee, @VisnuGanth, @jessibreen, @alaxalves, @kamal-kaur04, @sssash18, @chinmay021, @Srija616, @AnnaMinakova, @greenbrown, @taysa91, @TaishiRoy, @yagmurbaydogan, @Shulammite-Aso, @Jigyasa-Kumari

cesswairimu, 2020-06-30:
@keshav234156, @WaldenJosh, @B4Lee, @VisnuGanth, @jessibreen, @alaxalves, @kamal-kaur04, @sssash18, @chinmay021, @Srija616, @AnnaMinakova, @greenbrown, @taysa91, @TaishiRoy, @yagmurbaydogan, @Shulammite-Aso, @Jigyasa-Kumari @moliver4, @justinelai, @Snouzy, @stevenalila, @Digiox, @yandu31, @milosv1, @jamescapps, @Nevenem, @chirag-wadhwa, @yogeshjdeshmukh, @omizrahi99, @theOpetunde, @sailahonkanen, @robin-natale, @avidler, @ccyccyccy
@urvashigupta7, @daemon1024, @Tlazypanda, @jscastanos, @sarthak-choudhary, @dongskyler, @keshavsethi, @Shulammite-Aso, @ckchu8, @keshav234156, @DerekChristy, @Srija616, @priyanka-choubey, @AyanChoudhary, @imajit, @NitinBhasneria, @nstjean, @marientje, @rutger-t, @Janet33, @TroydonAnabolic, @priyanshbalyan, @Medlineme4, @mandysGit, @msanden, @AidenOliver, @Janetsomehow, @AntroMarina, @llabake, @khushi02, @Alexjake660, @LarrySul, @nicoleiocana, @chrisdrums1, @Angelvicks, @Akanksha1212, @beladinaelfitri, @rid1hady, @shreyaa-s, @goamoth, @SusannaDiV, @yusofsf, @nanlliu, @WaldenJosh, @B4Lee, @VisnuGanth, @jessibreen, @alaxalves, @kamal-kaur04 @adityakpranav @esiebomaj @Yonttee @swarajpure @vinuch @ishika1727

cesswairimu, 2020-06-30:
@Janet33, @TroydonAnabolic, @priyanshbalyan, @Medlineme4, @mandysGit, @msanden, @AidenOliver, @Janetsomehow, @AntroMarina, @llabake, @khushi02, @Alexjake660, @LarrySul, @nicoleiocana, @chrisdrums1, @Angelvicks, @Akanksha1212, @beladinaelfitri, @rid1hady, @shreyaa-s, @goamoth, @SusannaDiV, @yusofsf, @nanlliu, @WaldenJosh, @B4Lee, @VisnuGanth, @jessibreen, @alaxalves, @kamal-kaur04 @adityakpranav @esiebomaj @Yonttee @swarajpure @vinuch @ishika1727 @nsachin08 @Rohit-786 @daemon1024

cesswairimu, 2020-06-30:
To the interns and all contributors...If you are working on something and find a minor bug or after a completion of feature and you realize you missed smth minor on the implementation...kindly consider converting these to ftos and link them [here](https://github.com/publiclab/plots2/issues/7683)...or just label the issue as a `fto-candidate` and ping me...I will help convert it to a fto... please lets consider doing this so that we can keep being welcoming to new contributors and keep the community growing...thanks all in advance :smile:

Tlazypanda, 2020-06-30:
Hey everyone :wave: This week I am working on alternative text for images, made some changes to existing prs and plan to optimise performance on further api endpoints :v:  If anyone has time, please review https://github.com/publiclab/plots2/issues/8057 which is the documentation of performance and accessibility guide :sweat_smile:  

Have opened https://github.com/publiclab/plots2/issues/8094 https://github.com/publiclab/plots2/issues/8093 https://github.com/publiclab/plots2/issues/8092 ftos for new contributors. 

Welcoming new contributors @habeckmatt  @nsachin08 :clap: :balloon:  please feel free to work on these issues and ping in case you run into any issues :tada:

jywarren, 2020-06-30:
@Tlazypanda this is super -- thank you! Other @publiclab/soc team members -- creating `first-timers-only` issues is an important part of your projects, and of working on team leadership. Many of you have made them before; it's a great chance now to begin integrating the creation of new FTO issues into your weekly workflow, and to start building a broader team with the skills to contribute! 🤝 ❤️ 

Thanks @cesswairimu for the reminder on evals!!!

Sometimes if I don't have time to make a proper FTO, I mark things with `fto-candidate` and leave this text:

```
This has been marked as a good candidate for becoming a [first-timers-only issue](http://www.firsttimersonly.com) like [these](https://code.publiclab.org/#r=all), meaning that it's simple, self-contained, and with some extra formatting, could be a great entry point for a new contributor. If you're familiar enough with this code, please consider [reformatting or reposting it as a first-timers-only issue](https://publiclab.org/notes/warren/10-31-2016/create-a-welcoming-first-timers-only-issue-to-invite-new-software-contributors), and then ping @publiclab/reviewers to get it labelled. Or, if this is not your first time, try to solve it yourself!
```

It's actually a "saved reply" of mine 😄 

[image]

Fun tip for being a project maintainer ❤️

jywarren, 2020-06-30:
Also - i wanted to remind GSoC team members @NitinBhasneria @keshavsethi @keshav234156 and @alaxalves that the DIAL program has asked each of you to provide very brief weekly blog posts and "standing meeting" check-ins. You can see some of you have been doing this each week:

https://hub.osc.dial.community/t/weekly-gsoc-student-checklist/1640/2

But I believe these are required from everyone. Not to put @alaxalves on the spot 😄  but he's done a great job (https://hub.osc.dial.community/t/publiclab-mapknitter-export-module-and-spectral-workbench-upgrades/1610/9) and you can see the updates each week are quite short. Standups too, from @keshavsethi - https://hub.osc.dial.community/t/weekly-gsoc-standups-for-2020-w27/1754/11 - great job!

I'm sorry for the redundancy, but what would be really nice and could maybe save time is if you checked in here each week, and then posted that over on DIAL as well, so you don't have to write multiple updates. Could that work for folks? Thank you! DIAL has been very generous supporting us so I want to be sure we fulfill their expectations too!

jywarren, 2020-06-30:
Hi all - one more thing for @NitinBhasneria @keshavsethi @keshav234156 and @alaxalves -- as evals are due this week, if you could ensure that your project planning issues are up to date and even a brief (one sentence) recap/summary of where you are in your project would be super helpful to ensure we do a thorough job writing evals! Great work getting this far!!! 🎉

alaxalves, 2020-06-30:
Hey @jywarren 
Should I write this Evaluation 1 recap at publiclab.org?

keshavsethi, 2020-06-30:
Hey Everyone :wave: 
Last week I worked on the pagination of spam2 and refined its UI a bit. I have also added notification settings for Digest. Next week I am going to work on flag posts and queues and write their tests. 

**I have updated my planning issue with the list of tasks completed**
https://github.com/publiclab/plots2/issues/7885 

**This is the list of my major PRs**
 https://github.com/publiclab/plots2/pull/7969
https://github.com/publiclab/plots2/pull/8022
https://github.com/publiclab/plots2/pull/8063
https://github.com/publiclab/plots2/pull/8034
https://github.com/publiclab/plots2/pull/8038
https://github.com/publiclab/plots2/pull/8058

**All of my weekly blogs at https://publiclab.org are mentioned below**( or you can find them at https://publiclab.org/tag/keshav)
- **Week  3 Coding week 0**  https://publiclab.org/notes/keshav_sethi0004/05-24-2020/spam-management-dashboard-gsoc-week-3
- **week  5 Coding week 1**  https://publiclab.org/notes/keshav_sethi0004/06-07-2020/gsoc-2020-week-5-coding-period-week-1-spam-management-dashboard
- **week  6 Coding week 2**  https://publiclab.org/notes/keshav_sethi0004/06-16-2020/gsoc-2020-week-6-coding-period-week-2-spam-management-dashboard
- **week  7 Coding week 3**  https://publiclab.org/notes/keshav_sethi0004/06-29-2020/gsoc-2020-week-7-coding-period-week-3-spam-management-dashboard
- **week 8 Coding week 4**  https://publiclab.org/notes/keshav_sethi0004/06-29-2020/gsoc-2020-week-8-coding-period-week-4-spam-management-dashboard

**All of these weekly blogs can also be found at DIAL** https://hub.osc.dial.community/t/public-lab-spam-management-dashboard/1757

@jywarren I am writing Documentation for Digest and bulk moderation(spam2) which you have suggested. I will update you about this soon.

@jywarren @cesswairimu @VladimirMikulic @emilyashley @pydevsg @ananyaarun **and others** 
Thank you!!  :heart: :tada:

cesswairimu, 2020-07-06:
Moved to https://github.com/publiclab/plots2/issues/8113#issuecomment-654164982 . Thanks everyone :+1:

closed 2020-07-06 by cesswairimu