publiclab/spectral-workbench #541: Weekly Community Check-In #73 - GSoC Wrap up

Shulammite-Aso, 2020-08-24:
Hi everybody! 👏

We all at Public Lab 🎈 - learn, grow, work, brainstorm ideas, contribute together so why not share about our weekly goals and the awesome work we have done at Public Lab with each other, so we can support and collaborate with each other better. We have a Community Check-In each week, where every community member can share something about their work from the past week and about their current week's goal 🎯 . You are also welcome to share fun-fact 😄 , new ideas 💡 , your learning goals ☑️.

We believe in collaborative efforts to support our community. We are running a learning platform which helps a newcomer to become a leader of tomorrow. 💯

## GSoC wrap up.

We have officially gotten to the last week of Google Summer of Code internship. I remember  @keshav234156 @NitinBhasneria @keshavsethi @alaxalves are the participants for this round. :tada:  Let's use this week to talk about some of their works so far or something about them that we appreciate the most.   Lets also look out for ways we can help make wrapping up their internship projects smooth for them, they can be suggestions as well.😄
### For interns
If there are any issues you're working on this week and think you may need some extra help with, please share them with everyone here. Also share any concerns you may be having with wrapping up your project!
## Next Check-in
If you would like to open the next check-in..Leave a comment below. We are happy to help if its your first time

**Thanks everyone for making Public Lab awesome** 🎈
Have a great week and year ahead all ❤️

cesswairimu, 2020-08-24:
@jywarren,  @rexagod, @dependabot[bot], @NitinBhasneria, @VladimirMikulic, @keshav234156, @shreyaa-sharmaa, @Shulammite-Aso, @cesswairimu, @sagarpreet-chadha, @manansingh96, @IgorWilbert, @DivyeshPuri, @geekychasser, @shreyaa-s, @tsparksh, @ananyo2012, @anto004, @avi-jain, @man-ahmad, @djmgit, @DevanshCodes, @DiegoValenciaRoman, @gauravano, @ijeomaemeruwa, @spelgubbe, @okonek, @j-tanz, @namanrai, @nstjean, @nfarve, @Rishabh04-02, @poppashingles, @sidntrivedi012, @pythongiant, @ryzokuken, @Uzay-G, @vrk99, @Wesbest, @Yugandhartripathi, @az-chris, @cindychangy, @faithngetich, @kunal-mohta, @smohtadi @NitinBhasneria, @shreyaa-sharmaa, @keshav234156, @sagarpreet-chadha, @ijeomaemeruwa, @DiegoValenciaRoman, @dependabot[bot], @shreyaa-s, @Shulammite-Aso, @cesswairimu, @alaxalves

cesswairimu, 2020-08-24:
@Shokescodes, @avats-dev, @NC25, @arteevraina, @Tlazypanda, @jywarren, @Shulammite-Aso, @Mohitkumar6122, @Ganeshjadhav221b, @RatanNarayanHegde, @keshavsethi, @LifeofDan-EL, @NitinBhasneria, @dms-yondy, @siddhama, @renugasaraswathy, @AgyeyaMishra, @Jolios, @cesswairimu, @samarakoon-ryan, @Samasth, @malaikaaiyar, @OmLakshmi, @nsachin08, @galanggg, @ishika1727, @icarito, @RajatSablok, @mronauli

Shulammite-Aso, 2020-08-24:
HI all. It's been particularly great working with @keshav234156 and  @NitinBhasneria so far. Keep being awesome guys :smile: :tada: 

Was mostly unable to do a lot here last week. Became was tough and busy for me as my family moved to a different place and for a number of reasons i'm the main person who can help around to make sure things work fine. 
This week i'm going to continue working on the issues and PR i currently have open.

ebarry, 2020-08-24:
Glad to hear that you were able to make things work fine last week @Shulammite-Aso ❣️
Over here, New Orleans staff is dealing with a couple extreme storms, and some might have to evacuate. 
So lovely to chat with everyone here :)

Shulammite-Aso, 2020-08-25:
Thanks @ebarry 
Sad to hear about the storms :grimacing:

cesswairimu, 2020-08-25:
@ebarry please stay safe :heart: 
A reminder to GSoC interns, final evals are currently ongoing deadline is on 31st. Congrats to you all :tada:

Tlazypanda, 2020-08-25:
Hey everyone :wave: 

A round of applause for all the gsoc interns :100: :clap: :tada: This summer of code program has definitely brought all of us closer as a team and am looking forward to staying connected with you ahead in life as well :tada: 

@ebarry Please take care :heart: 

This week worked on some more performance optimisations https://github.com/publiclab/plots2/pull/8339 and print template refinement https://github.com/publiclab/plots2/pull/8337 would love to get your inputs on this :v:

jywarren, 2020-08-25:
Hi all - echoing @Shulammite-Aso @cesswairimu @Tlazypanda and others in congratulating the GSoC team (and one more congrats to the Outreachy team!)! 🎉 🎉 🎉 🎉 

I'm filing some last minute issues to help us get over the finish line, as well!

I know folks are already using a lot of the new features, so you should all be immensely proud of the work to date!

@Shulammite-Aso please take the time you need, no worries at all. Thanks for being in touch and for opening this week's check in!

Also, hoping you are all staying safe wherever you are ❤️

keshavsethi, 2020-08-25:
Hey Everyone 👋🏻 
The last 3 months of internship here at PL ❤️  were really _awesome_ :heart_eyes: and I learned a lot to things from everyone!! 🎉 🎉 

Thanks to all the mentors, community members, and everyone for helping, reviewing, and continuously supporting me throughout the GSoC period @jywarren @cesswairimu @VladimirMikulic @pydevsg @SidharthBansal @Uzay-G @ebarry and all the fellow members and interns. I am really looking forward  to contributing a lot in PL ❤️ 🎉 

@jywarren | **For Final evaluation**: What should I put in "Provide a link to the work you did during GSoC 2020 "?
Should I provide a link to all my PRs or something else? Please suggest

Last week I worked on the Insight section of spam2 and tests https://github.com/publiclab/plots2/pull/8289 (merged, thanks @jywarren ). Please refer to  documentation at https://publiclab.org/notes/keshav_sethi0004/08-18-2020/spam-management-dashboard-documentation

Thanks, everyone!! 🚀 🎉

LifeofDan-EL, 2020-08-25:
Hey, everyone, I'm kind of new here. I'm an Android developer with technical skills. I just looking for a way to contribute to the community. I'm quite good with docs and hopefully, they will be an Android project idea for GSoC next year.

ebarry, 2020-08-27:
Welcome @LifeofDan-EL !

For those here finishing up summer fellowships, thank you for writing this great documentation and please keep it on hand as you prepare for the video presentation on September 14. Please hold your calendars to be available that day! See the time on the calendar at https://publiclab.org/events -- scroll down a bit

sagarpreet-chadha, 2020-09-03:
Moved to https://github.com/publiclab/plots2/issues/8374

closed 2020-09-03 by sagarpreet-chadha