publiclab/spectral-workbench #608: Support for iOS devices and Safari on MacOS for live camera feed

waridrox, 2021-03-23:
**Please describe the desired behavior.**

- Currently on (iOS 14/ iPadOS) in safari or chrome browser, the live camera feed cannot be loaded both on _https://spectralworkbench.org/capture_ or on localhost:8080/examples/capture/. This means that iOS users are out of luck since these popular browsers can't support the live feed. 

- On safari for MacOS there's a different scenario observed where on localhost, live feed does load after prompting for camera access. But at _https://spectralworkbench.org/capture_ it doesn't. 

___
**Additional context (optional)**

_Tested on iPhone simulator 1 at localhost_
[image]

___
_Tested on iPhone simulator 2 at https://spectralworkbench.org/capture_
[image]
___
_Tested on iPadOS on Safari at https://spectralworkbench.org/capture_

https://user-images.githubusercontent.com/58583793/112166047-4b45ce80-8c15-11eb-9af0-472ef2d4c96d.MP4

jywarren, 2021-03-30:
This should be achievable with the new Capture code in https://github.com/publiclab/spectral-workbench.js/issues/219, and so we should note that substantial project would solve this and many other issues in this repository. Thank you, it's a great suggestion!!!

jywarren, 2021-09-23:
This may be complete now!

jywarren, 2021-09-23:
Aha - it still gets stuck on the initial frame -- it connects to camera on iOS but freezes immediately. Should debug with an iOS device!

jywarren, 2021-09-23:
https://github.com/publiclab/spectral-workbench.js/issues/266