publiclab/spectral-workbench #638: 500 error on saving new spectrum

icarito, 2021-05-31:
After deploying I wanted to test saving a new spectrum and am getting:
```
web_1  | I, [2021-05-31T16:46:22.066689 #12512]  INFO -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c] Command :: file -b --mime '/tmp/d7ba9bbfda42b9657f14ee37ef76150b20210531-12512-10n
tqof.png'
web_1  | D, [2021-05-31T16:46:22.587425 #12512] DEBUG -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c]   Spectrum Create (252.2ms)  INSERT INTO `spectrums` (`title`, `author`, `notes`, 
`photo_file_name`, `photo_content_type`, `photo_file_size`, `created_at`, `updated_at`, `user_id`) VALUES ('Test', 'icarito', 'Short test', 'capture.png', 'image/png', '143198', '2
021-05-31 16:46:22', '2021-05-31 16:46:22', 6329)
web_1  | D, [2021-05-31T16:46:22.653996 #12512] DEBUG -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c]    (63.5ms)  ROLLBACK
web_1  | I, [2021-05-31T16:46:22.654971 #12512]  INFO -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c] Completed 500 Internal Server Error in 4628ms (ActiveRecord: 319.0ms)
web_1  | F, [2021-05-31T16:46:22.656939 #12512] FATAL -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c]
web_1  | F, [2021-05-31T16:46:22.657057 #12512] FATAL -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c] Errno::EEXIST (File exists @ dir_s_mkdir - /app/public/system):
web_1  | F, [2021-05-31T16:46:22.657110 #12512] FATAL -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c]
web_1  | F, [2021-05-31T16:46:22.657170 #12512] FATAL -- : [74bad2ee-32fd-46c3-9d6d-6d2eec2d597c] app/controllers/spectrums_controller.rb:230:in `create'
```

icarito, 2021-05-31:
This issue has been tracked down to a misconfigured disk volume mount in the container.
After correct configuration, the spectrum is created fine, but there appears a 500 error:
```
web_1  | I, [2021-05-31T17:22:11.790425 #133]  INFO -- : [f15b54e8-2433-450a-9663-426f983db94e] Completed 500 Internal Server Error in 420ms (ActiveRecord: 54.0ms)                 
web_1  | F, [2021-05-31T17:22:11.792439 #133] FATAL -- : [f15b54e8-2433-450a-9663-426f983db94e]                                                                                     
web_1  | F, [2021-05-31T17:22:11.792623 #133] FATAL -- : [f15b54e8-2433-450a-9663-426f983db94e] ActiveRecord::NotNullViolation (Mysql2::Error: Column 'lat' cannot be null: UPDATE
```
Seems to be some kind of validation issue.

jywarren, 2021-05-31:
Could be this: https://stackoverflow.com/a/48120366/1116657

jywarren, 2021-05-31:
OK when i save a live captured spectrum i'm seeing:

```
geotag: false
lat: 
lon: 
```

jywarren, 2021-05-31:
Was there a change in policy where `null: false` is now by default in Rails 4?

jywarren, 2021-05-31:
OK still not sure how this happened, but production had `null: false` i.e. prohibited, while GitPod showed `null: true` for the lat and lon columns, meaning allowed. How did production get this way? We are manually changing it but... you'd think it'd have been an issue before. Have we migrated the whole db and how could such a setting not get migrated too?

jywarren, 2021-05-31:
There do seem to be some changes in defaults behaviors between MySQL 5.6-5.7: https://github.com/rails/rails/issues/38622 (and others)

jywarren, 2021-05-31:
Gitpod is using: `mysql  Ver 8.0.21-0ubuntu0.20.04.4 for Linux on x86_64 ((Ubuntu)`

Production is using MariaDB, not sure what version, but probably post equivalent to MySQL 5.7.

jywarren, 2021-05-31:
https://github.com/publiclab/spectral-workbench/blob/b764f35429f00879c61771bd49199ccf65ae2c91/app/controllers/spectrums_controller.rb#L269-L276 are the lines

Aha!!!

[image]

jywarren, 2021-05-31:
so, strangely, gitpod lacks the limitation because it used all the migrations, but production skips them because it's supposed to already be in that state

https://github.com/publiclab/spectral-workbench/pull/499/files#diff-99e8bae2fa8f313d8a18aa9126557c29f965bb84f3c74786ac8a6403aee212e1L3

jywarren, 2021-05-31:
anyhow i think it's OK that we are doing it manually, as a result. By editing this file, Alex kind of changed history

jywarren, 2021-05-31:
So what I don't get is why the default of 0 wasn't used. We may need to go in and add default 0 to all records. We could possibly then re-add the `null: false` requirement? But not sure.

jywarren, 2021-05-31:
OK with the manual db change we are now able to save spectra from live capture! https://spectralworkbench.org/spectrums/220649

Leaving this open for follow-up in case needed... will re-assess tomorrow! Thanks @icarito

icarito, 2021-05-31:
Thank you Jeff for helping to debug the issue and confirming the table
alteration at the live database!
--
Sebastian Silva

closed 2021-06-04 by icarito