publiclab/spectral-workbench #640: I believe I know why SpectralWorkbench is having some issues...

RoyceThomas, 2021-05-31:
Hello Public Lab or whoever this may concern, I am a representative of Penrith Public High School, NSW Australia, and I believe the reason why the site is having some issues is because of us. A few weeks ago, we were tasked to use your website as a key commodity to a HSC assignment procedure. So, 60 students were simultaneously using your website to upload, edit, refine graphs of spectrographs. Fortunately, this assignment is due tomorrow morning, so 1st of June so you'll be happy to know that most of our students would not be using the website after tomorrow and you can get back to normal functionality. I however, would continue using the website just to see what else can offer, other than just being a computer to show spectrograms. 
Thank you for your service and sorry for the inconvenience.
Royce

jywarren, 2021-05-31:
We'll be doing an in-depth analysis in https://github.com/publiclab/spectral-workbench/issues/638 if you'd like to follow along.

jywarren, 2021-05-31:
Hi we believe we have resolved the issue and saving now works:

https://github.com/publiclab/spectral-workbench/issues/638

We'll check in again tomorrow. We also observed an intermittent/unpredictable issue with spectrum upload at https://spectralworkbench.org/upload, tracking that in #641 -- more on that one soon. I'll close this for now but please tell us if you have more trouble!!

closed 2021-05-31 by jywarren