publiclab/spectral-workbench #641: /upload is erroring with 404 not found

jywarren, 2021-05-31:
https://spectralworkbench.org/upload

https://github.com/publiclab/spectral-workbench/blob/main/config/routes.rb#L40 - not sure why, the route looks good

jywarren, 2021-05-31:
Strange, @icarito doesn't see this:

[image]

jywarren, 2021-05-31:
Tested multiple times more and i'm still unable to access it from my chromebook. will try other devices. Sebastian still able to access it.

jywarren, 2021-05-31:
And it works fine in GitPod on my chromebook...

jywarren, 2021-05-31:
Same 404 from Android phone chrome browser:

[image]

norenTM, 2021-06-06:
Is this going to be addressed soon? I have a project due very soon and I'm worried that I will have incomplete data. I can't get the upload page to load on any device.

icarito, 2021-06-06:
Hello,
Yes after some discussion and looking at the feedback, we've decided to
rollback into the previous version tonight.
Please excuse the disruption in the regular functionality of Spectral
Workbench.
Regards,
--
Sebastian Silva
System Administrator at Public Lab

icarito, 2021-06-07:
We've rolled back to a previous version. Please let us know if you continue to have problems.

norenTM, 2021-06-07:
Logging in results in a 404 😅

norenTM, 2021-06-07:
> Attempting to log in with any service results in a 404 😅

This has been resolved now, I assume due to time. Thank you very much for your efforts, all 404-related bugs (/upload, downloading files) have all been eradicated.

jywarren, 2021-06-11:
Thank you for the extra checks @norenTM -- much appreciated. We have been simultaneously dealing with some very high bot activity on PublicLab.org and trying to mitigate downtime in relation to that.

For folks working on this issue, we rolled back but the error is still visible at https://stable.spectralworkbench.org/upload (our staging server) -- compare to https://spectralworkbench.org/upload which now works post-rollback. 

NOTE: sessions and cookies make logging into both from the same browser tough. Try logging into https://stable.spectralworkbench.org/upload from an incognito tab, and https://spectralworkbench.org/upload from a regular tab. 

I'm really mystified at what could be causing this. Will look at any possible routing changes that could possibly affect this. Also CC'ing @waridrox just for a heads up!

jywarren, 2021-06-11:
Could Bootsnap be involved here? It provides a lot of caching.

jywarren, 2021-06-11:
OK seems the big upgrade in #539  is likely to be the issue - it removed default routing at:

[image]

https://github.com/publiclab/spectral-workbench/commit/98dbde7a881788f7bd48db59eee616d6de70054b#diff-959bc9abc46a55332bb64d5155a79323afa75a50ec1a2137ddd22d926f62c6c5L193

I bet we had really been relying on it. We can use GitPod to ensure routing works correctly and that should resolve this. 

Any upload issues are likely to have a similar issue!

jywarren, 2021-06-15:
OK, update -- i wasn't able to reproduce this in gitpod -- it shows upload fine:

[image]

Looking deeper!

jywarren, 2021-06-15:
OK, on stable, I can see it happening. And I think we may be GETTING a 500 error, but for some reason be SEEING a 404. I confirmed it's a 404 in the Chrome console. But, on the server side, i see:

```
I, [2021-06-15T22:10:48.491057 #349]  INFO -- : [e7fafce2-2dc2-436d-89e6-68c583d81957] Redirected to https://stable.spectralworkbench.org/upload
I, [2021-06-15T22:10:48.491247 #349]  INFO -- : [e7fafce2-2dc2-436d-89e6-68c583d81957] Completed 302 Found in 34ms (ActiveRecord: 29.6ms)
I, [2021-06-15T22:10:48.516115 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4] Started GET "/upload" for 72.92.236.166 at 2021-06-15 22:10:48 +0000
I, [2021-06-15T22:10:48.517812 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4] Processing by SpectrumsController#new as HTML
D, [2021-06-15T22:10:48.520521 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mUser Load (0.8ms)ESC[0m  ESC[1mESC[34mSELECT  `users`.* FROM `users` WHERE `users`.`id` = 1 LIMIT 1ESC[0m
D, [2021-06-15T22:10:48.521696 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mCACHE User Load (0.2ms)ESC[0m  ESC[1mESC[34mSELECT  `users`.* FROM `users` WHERE `users`.`id` = 1 LIMIT 1ESC[0m  [["id", 1], ["LIMIT", 1]]
I, [2021-06-15T22:10:48.524874 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   Rendering layouts/application.html.erb
I, [2021-06-15T22:10:48.525478 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   Rendering spectrums/new.html.erb within layouts/application
I, [2021-06-15T22:10:48.530425 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   Rendered spectrums/_new_image.html.erb (0.7ms)
D, [2021-06-15T22:10:48.531328 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mCACHE User Load (0.2ms)ESC[0m  ESC[1mESC[34mSELECT  `users`.* FROM `users` WHERE `users`.`id` = 1 LIMIT 1ESC[0m  [["id", 1], ["LIMIT", 1]]
D, [2021-06-15T22:10:48.532618 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mCACHE User Load (0.3ms)ESC[0m  ESC[1mESC[34mSELECT  `users`.* FROM `users` WHERE `users`.`id` = 1 LIMIT 1ESC[0m  [["id", 1], ["LIMIT", 1]]
D, [2021-06-15T22:10:48.589230 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mSpectrum Load (19.0ms)ESC[0m  ESC[1mESC[34mSELECT spectrums.id, spectrums.created_at, spectrums.user_id FROM `spectrums` INNER JOIN `tags` ON `tags`.`spectrum_id` = `spectrums`.`id` WHERE `spectrums`.`user_id` = 1 AND (tags.name = ('calibration') OR tags.name LIKE ('linearCalibration:%')) ORDER BY spectrums.created_at DESCESC[0m
D, [2021-06-15T22:10:48.789492 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mSpectrum Load (189.3ms)ESC[0m  ESC[1mESC[34mSELECT `spectrums`.* FROM `spectrums` INNER JOIN `tags` ON `tags`.`spectrum_id` = `spectrums`.`id` WHERE `spectrums`.`id` IN (120774, 112148, 92538, 89900, 89557, 88735, 88734, 88733, 88732, 88731, 88730, 88729, 88728, 85787, 83326, 82252, 82252, 82251, 79087, 73760, 72806, 72366, 72365, 72364, 72360, 71686, 71243, 71081, 71075, 70824, 70808, 70790, 70771, 70770, 70769, 70768, 70723, 70723, 70695, 69973, 69704, 69703, 69662, 69662, 68681, 68680, 68588, 68585, 68577, 66713, 66713, 63935, 59744, 59744, 58942, 58236, 55077, 55067, 38256, 36143, 36006, 16070, 16069, 14917, 14182, 12305, 12106, 12067, 12065, 10863, 10219, 9590, 8443, 8427, 8426, 8425, 8424, 8423, 8421, 8420, 8418, 8416, 8000, 7999, 7998, 7957, 7950, 7949, 7948, 5708, 5162, 5161, 5160, 5159, 5158, 5157, 5156, 5155, 5154, 5153, 5152, 5151, 5150, 5148, 5147, 5146, 5145, 5144, 5143, 5142, 5130, 5129, 5128, 5127, 5126, 5125, 5124, 4780, 4778, 4777, 4776, 4772, 4769, 4768, 4767, 4766, 4763, 4762, 4737, 4735, 4734, 4732, 4731, 4729, 4726, 4717, 4711, 4705, 4700, 4699, 4690, 4631, 4520, 4038, 1980, 1739, 1714, 1390, 1175, 1033, 800, 800, 707, 707, 707, 596, 595, 595, 591, 555, 555, 551, 551, 482, 482, 475, 442, 442, 442, 413, 413, 413, 408, 408, 408, 408, 384, 377, 374, 371, 364, 351, 351, 333, 320, 101, 46, 45, 22, 21, 21)ESC[0m
D, [2021-06-15T22:10:48.918450 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mCACHE User Load (0.1ms)ESC[0m  ESC[1mESC[34mSELECT  `users`.* FROM `users` WHERE `users`.`id` = 1 LIMIT 1ESC[0m  [["id", 1], ["LIMIT", 1]]
D, [2021-06-15T22:10:48.922459 #349] DEBUG -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   ESC[1mESC[36mTag Load (0.6ms)ESC[0m  ESC[1mESC[34mSELECT `tags`.* FROM `tags` WHERE `tags`.`id` IN (NULL, NULL)ESC[0m
I, [2021-06-15T22:10:48.923419 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   Rendered spectrums/new.html.erb within layouts/application (397.8ms)
I, [2021-06-15T22:10:48.923523 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   Rendered layouts/application.html.erb (398.5ms)
I, [2021-06-15T22:10:48.923834 #349]  INFO -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4] Completed 500 Internal Server Error in 406ms (ActiveRecord: 212.3ms)
F, [2021-06-15T22:10:48.924948 #349] FATAL -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   
F, [2021-06-15T22:10:48.925007 #349] FATAL -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4] ActiveRecord::RecordNotFound (Couldn't find all Tags with 'id': (all, {:conditions=>{:user_id=>1, :name=>"calibration"}, :include=>:spectrum, :limit=>20, :order=>"id DESC"}) (found 0 results, but was looking for 2).):
F, [2021-06-15T22:10:48.925049 #349] FATAL -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4]   
F, [2021-06-15T22:10:48.925109 #349] FATAL -- : [f3ff1510-fdd5-47a9-b781-baa27dd36dc4] app/models/user.rb:100:in `tag'
[f3ff1510-fdd5-47a9-b781-baa27dd36dc4] app/views/spectrums/new.html.erb:113:in `_app_views_spectrums_new_html_erb__2650622626532582424_70304386915120'
```

jywarren, 2021-06-15:
Key section:

```
Completed 500 Internal Server Error in 406ms (ActiveRecord: 212.3ms)
ActiveRecord::RecordNotFound (Couldn't find all Tags with 'id': (all, {:conditions=>{:user_id=>1, :name=>"calibration"}, :include=>:spectrum, :limit=>20, :order=>"id DESC"}) (found 0 results, but was looking for 2).):
app/models/user.rb:100:in `tag'
```

jywarren, 2021-06-15:
https://github.com/publiclab/spectral-workbench/blob/afe7c986bbdd871d76ec162dd1d8896b2a165f07/app/models/user.rb#L98-L104

jywarren, 2021-06-15:
I bet we can't reproduce in GitPod because it's database dependent. Like, recent calibrations or something!

jywarren, 2021-06-15:
And, we should figure out why we're seeing a 404 instead of a 500!

jywarren, 2021-06-15:
Yes, reproducing this will require having a spectrum tagged "calibration" in your profile! The fix, i hope, is at #651.

PR #651 (2021-06-15): fix tags query to resolve /upload 404/500 error
fixes #641

Make sure these boxes are checked before your pull request is ready to be reviewed and merged. Thanks!

* [ ] tests pass -- `rake test`
* [ ] code is in uniquely-named feature branch, and has been rebased on top of latest master (especially if you've been asked to make additional changes)
* [ ] pull request are descriptively named
* [ ] if possible, multiple commits squashed if they're smaller changes
* [ ] reviewed/confirmed/tested by another contributor or maintainer

Please be sure you've reviewed our contribution guidelines at htt [... cut]
commit 843e231f (2021-06-15): fix tags query to resolve /upload 404/500 error (#651)

fixes #641
closed 2021-06-15 by jywarren