publiclab/spectral-workbench #644: Upload URL is returning 404

icarito, 2021-06-04:
These URLs are expected to exist:

https://spectralworkbench.org/upload

https://spectralworkbench.org/spectrums/latest/220369.csv

icarito, 2021-06-04:
I can actually reach https://spectralworkbench.org/upload

norenTM, 2021-06-04:
When logged in, I'm not able to reach https://spectralworkbench.org/upload - have also tested this on different browsers.

jywarren, 2021-06-12:
Linking to #641

jywarren, 2021-06-15:
https://stable.spectralworkbench.org/spectrums/latest/220369.csv

jywarren, 2021-06-15:
We need to check if other pages are erroring, besides these. Wasn't there still an error saving?

jywarren, 2021-06-15:
https://stable.spectralworkbench.org/spectrums/latest/220369.csv is a routing error:

```
Started GET "/spectrums/latest/220369.csv" for 72.92.236.166 at 2021-06-15 22:13:15 +0000
ActionController::RoutingError (No route matches [GET] "/spectrums/latest/220369.csv"):
```

jywarren, 2021-06-15:
We'll add a route for CSVs (probably `params[:format]` thing) and check in GitPod.

closed 2021-06-15 by jywarren

jywarren, 2021-06-15:
we should watch for similar ones in case other routes or queries need fixing!!

commit b0be85e1 (2021-06-15): fixed csv routes, fixes #644