publiclab/spectral-workbench #659: Update .gitignore to ignore node_modules directory

waridrox, 2021-06-20:
Hi, this is a [first-timers-only issue](https://publiclab.github.io/community-toolbox/#r=all). This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [help wanted](https://github.com/publiclab/plots2/labels/help-wanted) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### About the `.gitignore` file 
The `.gitignore` file in the project directory is created to ignore all the unwanted files which would otherwise be tracked by git (a version control system). Since the addition of `package.json` file in our codebase will create various files under the `node_modules` folder, we would like them to be ignored.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 📝 **Update** the file [$FILENAME]($BRANCH_URL) in the `$REPO` repository (press the little pen Icon) and edit the line as shown below. 

[See this page](https://publiclab.github.io/community-toolbox/#r=all) for some help in taking your first steps!

Below is a "diff" showing in red (and a `-`) which lines to remove, and in green (and a `+`) which lines to add:
Add the following to the `packages.json` file (without the `+` sign):

```diff
+ node_modules
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:

1. If you are familiar with the terminal or would like to learn it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

2. You can also [edit files directly in your browser](https://help.github.com/articles/editing-files-in-your-repository/) and open a pull request from there. 

- [ ] 🏁 **Done** Ask in comments for a review :)

### 🤔❓ Questions?

Leave a comment below!

### Is someone else already working on this?

We encourage you to link to this issue by mentioning the issue # in your pull request, so we can see if someone's already started on it. **If someone seem stuck, offer them some help!** Otherwise, [take a look at some other issues you can help with](https://publiclab.github.io/community-toolbox/#r=all). Thanks!

(This issue was created by [First-Timers-Bot](https://github.com/hoodiehq/first-timers-bot).)

saraswathy-krish, 2021-06-20:
Hello @waridrox , I am a first-time contributor here and looking for a first-timers-only issue. Can I take this up?

waridrox, 2021-06-20:
Sure @saraswathy-krish! Go for it!🚀 and don’t hesitate to ask for help if required.

saraswathy-krish, 2021-06-20:
Thank you @waridrox! I will get started on it.

saraswathy-krish, 2021-06-21:
Hi @waridrox , I just now created a pull request. Please let me know in case of any issues. Thanks!

waridrox, 2021-06-21:
Hey @saraswathy-krish, just add the issue number to the PR, rest all LGTM 🎈

PR #662 (2021-06-22): Updated .gitignore to remove node_modules folder
Fixes #659 Updated .gitignore to remove the node_modules folder

Make sure these boxes are checked before your pull request is ready to be reviewed and merged. Thanks!

* [ ] tests pass -- `rake test`
* [ ] code is in uniquely-named feature branch, and has been rebased on top of latest master (especially if you've been asked to make additional changes)
* [ ] pull request are descriptively named
* [ ] if possible, multiple commits squashed if they're smaller changes
* [ ] reviewed/confirmed/tested by another contributor or maintainer

Please be s [... cut]
closed 2021-06-22 by jywarren