publiclab/spectral-workbench #673: Protect critical interfaces (/capture, /capture/v2, /upload, /spectrums/0000) with system tests

jywarren, 2021-07-03:
https://github.com/publiclab/spectral-workbench/pull/660 introduced a new page - let's protect it with some system tests: 

https://github.com/publiclab/spectral-workbench/tree/main/test/system

what can we brainstorm that we should test for?

We can similarly improve our confidence and speed merging other things if we protect other critical systems with system tests.

unnati914, 2021-07-04:
could you guide me to this?

waridrox, 2021-07-04:
I think the tests for `capture/v2` would make more sense once we implement the styling for that page. And that way we are able to target specific elements on the webpage in the tests.

As a result, other pages could be looked into before like `/capture` `/upload` and `/spectrums/0000`. 

Since things like uploading and going to the specific spectrum would require logging in to the app. We could test from simply going to the webpage...then logging in the app with some admin account...Capturing the spectra and saving the data at /spectrums/0000

The upload test could be carried out similarly, difference being that the image could be uploaded instead of captured which would also require user login. 

@jywarren, thoughts ?

waridrox, 2021-07-04:
Hi @unnati914 🙌 if you would like to contribute you might want to first install the software on your local machine and take a look at how plots2 tests are written - https://github.com/publiclab/plots2/tree/main/test/system. This might give you some idea how system tests work...

unnati914, 2021-07-06:
Hi thank you so much for the help :)

waridrox, 2021-07-20:
- Possible way of testing live capture / graph - using a GIF image / short vid clip for the spectrum. 
- Is BrowserStack a viable option for cross browser testing / can we use puppeteer.

unnati914, 2021-07-21:
Even I had thought of implementing this only 😄

waridrox, 2021-08-22:
`/capture`, `/capture/v2` and `/upload` implemented in #702