publiclab/spectral-workbench #710: not able to get the fake camera feed working w arguments: `--use-fake-device-for-media-stream` and `--use-fake-ui-for-media-stream` in the Selenium Chrome driver

jywarren, 2021-08-22:
Hi @jywarren, I'm not able to get the fake camera feed work 😕 by passing in the arguments: `--use-fake-device-for-media-stream` and `--use-fake-ui-for-media-stream` in the Selenium Chrome driver. Other than that I've created tests for `/upload`, `/capture` and `/capture/v2` routes which capture the spectrum and delete it after logging in with a random username using the `faker` gem.

More on this at - 

- https://webrtc.github.io/webrtc-org/testing/
- https://elh.mx/generales/testing-webrtc-fake-camera-with-cucumbercapybara-in-chrome/

_Originally posted by @waridrox in https://github.com/publiclab/spectral-workbench/issues/702#issuecomment-902173008_

jywarren, 2021-08-22:
Just moving here so we can keep working on this issue!

waridrox, 2021-08-22:
Here's the `fake-media-stream` that should show up for reference - 

[image]

jywarren, 2021-08-22:
For reference linking to https://github.com/publiclab/spectral-workbench.js/pull/245

And, i think the format for using a local test video is: 

https://github.com/webrtc/samples/issues/616#issuecomment-130722538

`--use-fake-device-for-media-stream --use-file-for-fake-video-capture="D:\Test\Video\y4m.y4m"`

?

jywarren, 2021-08-22:
https://docs.cypress.io/api/plugins/browser-launch-api#Use-fake-video-for-webcam-testing

Should this help?

Thanks for this - it's exciting!!!

waridrox, 2021-08-22:
Yes this would do, it's just that currently I don't have that much experience with cypress testing, so I'll circle back here to write some more tests later...

jywarren, 2021-08-22:
OK, great. If you could try creating a PR with the cypress test, and even if it doesn't work, just open and point me at where it's going wrong, I can help debug!

Great!!!

jywarren, 2021-08-22:
So just btw, is this solved in https://github.com/publiclab/spectral-workbench.js/ (or esp. in https://github.com/publiclab/spectral-workbench.js/pull/245)? and will solving it there solve it here too, or is there a distinct issue with this argument in Rails system tests?

jywarren, 2021-09-21:
I believe we were able to resolve #245 without actually faking the camera input - we just checked for a large enough file size in the downloaded image. Later if we add a video or still image of a spectrum, we could refine our tests to look for more, and pursue this!