publiclab/spectral-workbench #720: fixing bundler version mismatch with containers

jywarren, 2021-09-14:
https://github.com/publiclab/spectral-workbench/pull/717

https://github.com/publiclab/plots2/issues/9871

```
gitpod /workspace/spectral-workbench $ bundle --version
Bundler version 1.17.3
```

icarito, 2021-09-21:
Yes, the issue is that when building the container, version 1.17.3 is installed, and so it fails to install Gems. Is it necessary to declare bundler version? I think it should work if we just remove those lines (but probably not best practice).

jywarren, 2021-09-21:
I think they're auto-generated by Bundler so that they'll keep getting re-added. And i wonder if there are some breaking changes in bundler syntax? Even if we're not encountering them they did change a major version number.

jywarren, 2021-09-21:
Resolving in https://github.com/publiclab/spectral-workbench/pull/721 - thanks!

closed 2021-09-21 by icarito