publiclab/spectral-workbench #736: Add tests for routes to protect them

jywarren, 2021-10-05:
Based on this PR: https://github.com/publiclab/plots2/pull/10242

we can protect routes; a lot of routes were broken in a recent upgrade: https://github.com/publiclab/spectral-workbench/search?q=route&type=issues

Let's make a big list of them - you can see recently fixed ones in the "blame" view of https://github.com/publiclab/spectral-workbench/blame/main/config/routes.rb - and create first-timers-only issues to add route tests for them! Since they're tests, they'll be really easy to review and approve.

jywarren, 2021-10-05:
The format for a routes test is:

```ruby
	test "test user_sessions create route" do
	    assert_routing({ path: '/user_sessions', method: :post }, { controller: 'user_sessions', action: 'create' })
	end
```

Tlazypanda, 2021-10-06:
Ohh this is a great idea! @jywarren Will open some ftos based off this and possibly an issue in plots2 as well to get contributors to this repo :v:

mskalisz, 2021-12-02:
Hi! I've completed a FTO issue - I'm still pretty new to contributing to open source but I'm wondering if there is anything I can do to help with this issue? Is the list complete, or can I help to create some more FTO issues? Thank you!

Tlazypanda, 2021-12-21:
Hey @mskalisz Feel free to open more FTO issues based on this :100: . I usually look out for the routes that are recently added/modified for the tests?