publiclab/spectral-workbench #738: Search query optimization needed for some strings

jywarren, 2021-10-05:
While https://spectralworkbench.org/search/test returns results fine, the following seems to crash? the server to a 502 error:

https://spectralworkbench.org/search/calcium%20hydroxide

(don't test it, i did already)

Could maybe be tested at: 

https://stable.spectralworkbench.org/search/calcium%20hydroxide

(it works there!!!)

[image]

[image]

jywarren, 2021-11-09:
I noticed we can search for https://stable.spectralworkbench.org/search/calcium - so that's a workaround for now.

jywarren, 2021-11-09:
> Update: I've noticed this general pattern: when I search for "calcium chloride" (without quotes), I get the error. When I only search calcium, I get several pages of results; but attempting to "next" to the next page results in the error, so I can never get past the first page of results.

jywarren, 2021-11-09:
We should test this in GitPod.

RaviAnand111, 2022-01-24:
Hey @jywarren , I tried to test it and I think whats happening is, whenever something is searched which is not in database, instead of showing `Not Found` it gives error `502 Bad Gateway`. 
This is what I found, what do you think ?