publiclab/spectral-workbench #779: Add routes test for match/:id

Tlazypanda, 2021-11-05:
Hi, this is a [first-timers-only issue](https://publiclab.github.io/community-toolbox/#r=all). This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [help wanted](https://github.com/publiclab/plots2/labels/help-wanted) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 📝 **Update** the file [routes_test.rb](https://github.com/publiclab/spectral-workbench/blob/main/test/integration/routes_test.rb) in the `spectral-workbench` repository (press the little pen Icon) and edit the line as shown below. 

[See this page](https://publiclab.github.io/community-toolbox/#r=all) for some help in taking your first steps!

Below is a "diff" showing in read (and a `-`) which lines to remove, and in green (and a `+`) which lines to add:

```diff
+  test "test get request for match id" do
+    assert_routing({ path: '/match/:id', method: 'get' }, {controller: 'match', action: 'index', id: ':id' })
+  end
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:

1. If you are familiar with the terminal or would like to learn it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

2. You can also [edit files directly in your browser](https://help.github.com/articles/editing-files-in-your-repository/) and open a pull request from there. 

- [ ] 🏁 **Done** Ask in comments for a review :)

### 🤔❓ Questions?

Leave a comment below!

### Is someone else already working on this?

We encourage you to link to this issue by mentioning the issue # in your pull request, so we can see if someone's already started on it. **If someone seem stuck, offer them some help!** Otherwise, [take a look at some other issues you can help with](https://publiclab.github.io/community-toolbox/#r=all). Thanks!

(This issue was created by [First-Timers-Bot](https://github.com/hoodiehq/first-timers-bot).)

najmieh, 2021-11-28:
Hi I'd like to work on this issue.

Tlazypanda, 2021-12-02:
Hey @najmieh I can see you have already completd a first-timers-only issue. Do you want to give another first-timer a chance? :sweat_smile:

SteveSayantan, 2021-12-02:
Hello @Tlazypanda, may I have this issue assigned to me? I have been looking for an FTO issue for a long time 😄

Tlazypanda, 2021-12-03:
Hey @SteveSayantan go ahead!

SteveSayantan, 2021-12-07:
Thanks a lot @Tlazypanda

SteveSayantan, 2021-12-07:
https://github.com/publiclab/spectral-workbench/pull/800#issue-796557344
This is the commit, please give it a review and let me know of any changes in the comments.

SteveSayantan, 2021-12-13:
Hey @Tlazypanda would you please review my code and inform if anything is wrong with it? I am a newbie , please let me know if there is something I missed out.

Tlazypanda, 2021-12-21:
Closed via #800

closed 2021-12-21 by Tlazypanda