publiclab/spectral-workbench #801: [Planning] Rebuilding Spectral Workbench capture and analysis interface (Outreachy)

Georjane, 2021-12-07:
[ ] TASKS
Hi there 👋🏽  This issue is a draft of my internship tasks. I plan on contributing to this and much more as time unfolds!

[My Outreachy Proposal](https://publiclab.org/notes/pzgeorjane/11-02-2021/outreachy-proposal-rebuilding-spectral-workbench-capture-and-analysis-interface)

### Internship Tasks

- [ ] Locally install and run code by following installation steps on spectral-workbench [Readme](https://github.com/publiclab/spectral-workbench#installation) 
- [ ] Run tests and make sure everything is working fine

- [ ] Reconnect broken routes

- [ ] Copy the correct link from the contribution page and insert it on the Recent Spectra Section for the contributor's name that currently has a broken link
- [ ] Find broken routes on the spectral workbench app and add them to a list here

- [ ] Protect routes with functional tests

- [ ] Reproduce detection of fluorescent lights from legacy capture interface into new capture interface (this then offers to calibrate, with an alert along the top!)

- [ ] Debug Safari compatibility and confirm iOS compatibility

- [ ] Develop setting to increase capture image resolution above currently fixed 640px

- [ ] Reconnect and refine “Compare spectra” and “Find Similar” functions within the Rails app

- [ ] Prototype a “reduce noise with time averaging” function within the JS library

- [ ] Further develop a stand-alone program from viewer and capture modules without Rails app (for example, ability to drag + drop and view a graph for saved JSON data or image)

Georjane, 2021-12-07:
Hi, @jywarren @Tlazypanda @TildaDares this is a draft of my task. I will be editing it as time goes by. I would love to hear your point of view on this first draft.
For week 1, I want to complete onboarding which entails reading necessary materials on Public Lab like the google docs Jeff shared during the meeting, setting up the codebase, and familiarizing myself with the tests.

jywarren, 2021-12-07:
Thank you and this looks like a great start! Will you be getting the
JavaScript lib running separately first before trying it within the rails
app, or the other way around? Either would work!

Georjane, 2021-12-07:
> Thank you and this looks like a great start! Will you be getting the JavaScript lib running separately first before trying it within the rails app, or the other way around? Either would work!
> […](#)

I would love to get the js library running separately first. Let me add that as a task for my onboarding too. Thank you 💯

TildaDares, 2021-12-07:
Great job @Georjane! Your first draft looks good🚀

jywarren, 2021-12-07:
Ah, also! I wanted to say that for testing yours, we can send you a papercraft spectrometer that's designed to work with this software! If you want to email me your mailing address. It's also available to print out and cut out yourself (using a spare blank DVD-R) from https://publiclab.org/paper

We can try mailing you one that uses a webcam too - but it is a package, instead of an envelope, so it could take a while to arrive.

Georjane, 2021-12-08:
> Ah, also! I wanted to say that for testing yours, we can send you a papercraft spectrometer that's designed to work with this software! If you want to email me your mailing address. It's also [...]

I can print it out and cut it myself. It would be fun experiencing the DIY practical part of it. Thanks Jeff. I will add it to my tasks and keep you updated on my progress with it.

jywarren, 2021-12-09:
OK, just be sure to print it onto quite thick paper if possible, and I
believe there is a reverse side that prints all black -- but you can also
do that yourself with some black paint if it's easier on the printer.

Enjoy!

Tlazypanda, 2021-12-12:
Hey @Georjane This looks amazing!! :tada: I love the way the list is ordered to allow you to work in a gradual manner as you get deeper into the codebase. All the best Jane :heart: 

Adding some issues here for your reference -
- Reconnect broken routes https://github.com/publiclab/spectral-workbench/issues/737
- Protect routes with functional tests https://github.com/publiclab/spectral-workbench/issues/736 example issue - https://github.com/publiclab/spectral-workbench/issues/741

Georjane, 2021-12-13:
Okay, @Tlazypanda, Thank you very much, It's so kind of you to share the above links that will be very resourceful to me. Thank you

cesswairimu, 2022-02-25:
Hi @Georjane,  on the task you needed clarification on, I have created a discussion issue here  https://github.com/publiclab/spectral-workbench/issues/893 and I will add the info I can get and we can discuss there. Thanks