publiclab/spectral-workbench #803: Run tests and make sure everything is working fine

Georjane, 2021-12-08:


Georjane, 2021-12-08:
I have been debugging for the following tests that are failing after I set up the codebase

```
# Running:

........E

Error:
SpectrumTest#test_spectrum_creation:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    test/models/spectrum_test.rb:14:in `block in <class:SpectrumTest>'

bin/rails test test/models/spectrum_test.rb:6

...E

Error:
SpectrumTest#test_spectrum_clone,_copies_tags_&_snapshots:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    test/models/spectrum_test.rb:113:in `block in <class:SpectrumTest>'

bin/rails test test/models/spectrum_test.rb:111

........E

Error:
SpectrumsControllerTest#test_should_clone_spectrum_calibration:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    test/controllers/spectrums_controller_test.rb:286:in `block in <class:SpectrumsControllerTest>'

bin/rails test test/controllers/spectrums_controller_test.rb:282

.E

Error:
SpectrumsControllerTest#test_should_extract_spectrum:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    test/controllers/spectrums_controller_test.rb:391:in `block in <class:SpectrumsControllerTest>'

bin/rails test test/controllers/spectrums_controller_test.rb:388

E

Error:
SpectrumsControllerTest#test_should_save_spectrum_if_vertical:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    app/controllers/spectrums_controller.rb:221:in `create'
    test/controllers/spectrums_controller_test.rb:340:in `block in <class:SpectrumsControllerTest>'

bin/rails test test/controllers/spectrums_controller_test.rb:338

......E

Error:
SpectrumsControllerTest#test_should_calibrate_spectrum:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    test/controllers/spectrums_controller_test.rb:274:in `block in <class:SpectrumsControllerTest>'

bin/rails test test/controllers/spectrums_controller_test.rb:270

E

Error:
SpectrumsControllerTest#test_should_set_sample_row_to_spectrum:
[... 125 lines cut ...]
SetsControllerTest#test_should_show_spectra_set_new_embed:
ActionView::Template::Error: Mysql2::Error: Expression #8 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'swbench_test.snapshots.id' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by: SELECT DISTINCT(spectrums.id), spectrums.title, spectrums.created_at, spectrums.id, spectrums.calibrated, spectrums.user_id, like_count, snapshots.id AS snapshot_id FROM `spectrums` INNER JOIN `spectra_sets_spectrums` ON `spectra_sets_spectrums`.`spectrum_id` = `spectrums`.`id` INNER JOIN `spectra_sets` ON `spectra_sets`.`id` = `spectra_sets_spectrums`.`spectra_set_id` LEFT OUTER JOIN snapshots ON snapshots.spectrum_id = spectrums.id WHERE (spectra_sets.id = 980190962) GROUP BY spectrums.id
    app/views/graph/_spectra.html.erb:12:in `_app_views_graph__spectra_html_erb___316050910742546116_47360285760100'
    app/views/embed/set.html.erb:21:in `_app_views_embed_set_html_erb___4201803490828008830_47360308116840'
    app/controllers/sets_controller.rb:68:in `embed2'
    test/controllers/sets_controller_test.rb:53:in `block in <class:SetsControllerTest>'

bin/rails test test/controllers/sets_controller_test.rb:52

...............................................E

Error:
CaptureControllerTest#test_should_save_a_spectrum:
Paperclip::Errors::CommandNotFoundError: Could not run the `identify` command. Please install ImageMagick.
    app/models/spectrum.rb:303:in `image_from_dataurl'
    app/controllers/capture_controller.rb:80:in `save'
    test/controllers/capture_controller_test.rb:61:in `block in <class:CaptureControllerTest>'

bin/rails test test/controllers/capture_controller_test.rb:59

...........

Finished in 52.441870s, 4.1570 runs/s, 12.0514 assertions/s.
218 runs, 632 assertions, 0 failures, 20 errors, 0 skips
```

TildaDares, 2021-12-09:
@Georjane I think you need to install `ImageMagick`

Georjane, 2021-12-10:
> @Georjane I think you need to install `ImageMagick`. I'm not sure what the `mysql2` errors are.

Okay @TildaDares 
I have made some progress debugging some of the errors. I will install ImageMagick as well and give you feedback. Thank you very much.

Georjane, 2021-12-10:
Yayyyy! 🎉🎉🎉  All tests are green now. Thanks @TildaDares 
```
# Running:

...............................................................................................................................................................DEPRECATION WARNING: The asset "spectralworkbench/SpectralWorkbench.js" is not present in the asset pipeline. Falling back to an asset that may be in the public folder.
This behavior is deprecated and will be removed.
To bypass the asset pipeline and preserve this behavior,
use the `skip_pipeline: true` option.
 (called from _app_views_layouts_embed_html_erb__1682375649123061037_47238766088220 at /home/jane/outreachy/spectral-workbench/app/views/layouts/embed.html.erb:12)
...........................................................

Finished in 40.933214s, 5.3257 runs/s, 17.1499 assertions/s.
218 runs, 702 assertions, 0 failures, 0 errors, 0 skips
```

I used this query, `SET GLOBAL sql_mode=(SELECT REPLACE(@@sql_mode,'ONLY_FULL_GROUP_BY',''));` from stack overflow to fix the `mysql2` errors. Thanks again!

closed 2021-12-10 by Georjane

jywarren, 2021-12-13:
This is awesome! Great work @Georjane !!!!!! 🎉 very satisfying, huh?

That `ONLY_FULL_GROUP_BY` is a common issue for our more complex Rails query. Glad you figured it out! MySQL changed a core feature after v5.7 and it causes this all over our systems. :-(