publiclab/spectral-workbench #805: Print out and cut the papercraft spectrometer (using a spare blank DVD-R) from here following Papercraft Spectrometer Intro Kit

Georjane, 2021-12-13:


Georjane, 2021-12-13:
@jywarren i was able to get the materials i will be needing for the papercraft today. I am looking forward to putting it all together 

[image]

Georjane, 2021-12-13:
> Wow!!! Love it! Enjoy and feel free to ask me questions as I was one of the designers :-)
> […](#)

Sure, I will! Thanks

Georjane, 2021-12-20:
Hi @jywarren @TildaDares @Tlazypanda 
I finally succeeded to build the paper craft spectrometer. The trickest part for me was peeling off the thin paper on the DVD😁
But I was glad I made it till the end.
I used the spectrometer to take the following image of the light bulb in my room when testing to see if I got it right.😅😅
I hope I did! I think I now have to test the spectrometer through the capture spectrum on the website. I am not sure exactly what extra thing I have to do there but I will try it out and put my questions here.

[image]

Georjane, 2021-12-20:
[image]

Georjane, 2021-12-20:
[image]

jywarren, 2021-12-21:
This is really lovely - a very clear spectrum - great work!!!

closed 2022-01-27 by Georjane