publiclab/spectral-workbench #825: login button not clickable on hover i.e. cursor isn't pointer.

khushalbhardwaj-0111, 2021-12-30:
## What happened just before the problem occurred
When hovering over the login button in the drop-down and navbar the login button does not show as clickable and shows as normal text.

## Relevant URLs

https://spectralworkbench.org/

## PublicLab.org username
username: khushalbhardwaj0111

Steps to see the issue:
- Go the https://spectralworkbench.org/
- Hover over the login button (if not logged in)
- It does not appear as a clickable button like the others, i.e, `cursor: pointer` is not there.
- Also, the the login option in the dropdown does not appear clickable as well like a button.

[image]

## Browser, version, and operating system
Mozilla Firefox `95.0.1` 64-bit, Ubuntu 21.10

For bug reports, fill out the above template; for feature requests, you can delete the template.

Kunal0220, 2022-01-07:
Hi @khushalbhardwaj-0111 , I think I can resolve this issue , Can you just give me your permission to go with it?

khushalbhardwaj-0111, 2022-01-07:
@Kunal0220 sure go ahead, I am not that familiar with frontend and ruby so won't be able to help though. You can ask for other folks contributing in the project to guide you through, I am a beginner to this project 😺.

khushalbhardwaj-0111, 2022-01-07:
@Tlazypanda could you plz assign this issue to @Kunal0220 .

Tlazypanda, 2022-01-07:
Hey @khushalbhardwaj-0111 nice catch! Assigning to @Kunal0220  I think the relevant file to be modified is the _header.html.erb file in app/layouts directory. Happy Opensourcing :blue_heart:

Kunal0220, 2022-01-07:
Thank You @Tlazypanda ,
I'll try to fix this ASAP , as I am new to Open Source , If i get stuck , I'll come back to this place for your guidance.

Kunal0220, 2022-01-08:
Hey @Tlazypanda  I have figured out the point where the code is breaking and have found the solution to it.

https://user-images.githubusercontent.com/71934168/148646977-ffe3cd64-eaa6-41ce-b611-719ed04feb0a.mp4

But the issue is that I am not very comfortable with RubyOnRails(ROR) , can you please help in figuring out that how do I have to integrate the html changes that I found in ROR?

Kunal0220, 2022-01-08:
BTW I have done these changes in the Console and not in the codebase!

Kunal0220, 2022-01-11:
@Tlazypanda I am looking to this ASAP , ThankYou :)