publiclab/spectral-workbench #833: Find broken routes on the spectral workbench app and add them to a list here

Georjane, 2022-01-11:


Georjane, 2022-01-11:
List of broken routes
- All author links **(pzgeorjane)** in the images below are broken
[image]
[image]
[image]

Georjane, 2022-01-11:
@TildaDares @jywarren @Tlazypanda 
I keep seeing the letter h like ` <%=h ` in front of opening tags like in the image below. I wish to find out if this is a bug or there is a meaning to it. I have seen this in a couple of places. Thank you. 🙏🏽 

_**app/views/macros/show.html.erb**_
[image]

TildaDares, 2022-01-11:
@Georjane `<%= h` is an alias for `html_escape`. It is used to escape HTML tags https://api.rubyonrails.org/classes/ERB/Util.html#method-c-h

Georjane, 2022-01-11:
> @Georjane `<%= h` is an alias for `html_escape`. It is used to escape HTML tags https://api.rubyonrails.org/classes/ERB/Util.html#method-c-h

oh, I see. it makes a lot of sense now. thank you @TildaDares

Georjane, 2022-02-09:
@Tlazypanda 
please what do we mean by looking at the blame view here,
you can see recently fixed ones in the "blame" view of https://github.com/publiclab/spectral-workbench/blame/main/config/routes.rb

closed 2022-03-10 by Georjane