publiclab/spectral-workbench #834: Protect routes with functional tests

Georjane, 2022-01-11:


Georjane, 2022-01-11:
I tried running the following test on the macros_controller but I did not exactly understand the response I was getting
```
# frozen_string_literal: true

require 'test_helper'

class MacrosControllerTest < ActionController::TestCase
  test 'should get macros index' do
    get :index
    assert_response :success
    # assert_not_nil assigns(:spectrums)
  end
end
```

TildaDares, 2022-01-11:
Hi Jane, what kind of responses did you get? 

I don't know if this helps you understand what macros do
https://github.com/publiclab/spectral-workbench/blob/abe48d023d8c113e9b8c30718932a630b99ec3fc/app/views/macros/_pane.html.erb#L5

TildaDares, 2022-01-11:
I also found this article on https://publiclab.org/wiki/spectral-workbench-api-v1#Macros on Public Lab. It's deprecated but it should give you a better sense oof what Macros are.

Georjane, 2022-01-11:
> Hi Jane, what kind of responses did you get?
> 
> I don't know if this helps you understand what macros do
> 
> https://github.com/publiclab/spectral-workbench/blob/abe48d023d8c113e9b8c30718932a630b99ec3fc/app/views/macros/_pane.html.erb#L5

I tried accessing the enpoint /macros locally but I was getting the following error
[image]

When I try accessing it on the live version I have this page
[image]

Georjane, 2022-01-11:
> I also found this article on https://publiclab.org/wiki/spectral-workbench-api-v1#Macros on Public Lab. It's deprecated but it should give you a better sense oof what Macros are.

ohhh I understand macros now. Thank you very much @TildaDares 
I think I understand the error now

Georjane, 2022-01-11:
Now able to create a macro but got the following error from the show method
[image]

TildaDares, 2022-01-11:
@Georjane I think `find_by_title` method is supposed to be passed one argument. You can try this format and see if it works
https://api.rubyonrails.org/classes/ActiveRecord/FinderMethods.html#method-i-find_by

Georjane, 2022-01-11:
> @Georjane I think `find_by_title` method is supposed to be passed one argument. You can try this format and see if it works https://api.rubyonrails.org/classes/ActiveRecord/FinderMethods.html#method-i-find_by

thanks for the link @TildaDares 
let me read more on this

Georjane, 2022-02-08:
@TildaDares thank you, @Tlazypanda helped me out with the following syntax that works
`@macro = Macro.find_by :title params[:id], conditions: { user_id: @user.id }, order: 'id DESC'`

PR #835 (2022-02-09): Protect routes with functional tests
Fixes #834

Make sure these boxes are checked before your pull request is ready to be reviewed and merged. Thanks!

* [ ] tests pass -- `rake test`
* [ ] code is in uniquely-named feature branch, and has been rebased on top of latest master (especially if you've been asked to make additional changes)
* [ ] pull request are descriptively named
* [ ] if possible, multiple commits squashed if they're smaller changes
* [ ] reviewed/confirmed/tested by another contributor or maintainer

Please be sure you've reviewed our contribution guidelines at https://public [... cut]
closed 2022-02-09 by jywarren