publiclab/spectral-workbench #848: Port Geocode tool back into new Capture interface, next steps

Georjane, 2022-01-14:


Georjane, 2022-01-18:
Hi @Tlazypanda @TildaDares @jywarren 👋🏽 

I was able to upload my first spectrum today locally. 
I was trying to test the functionality of the geocoding tool on the old capture interface to compare it with the new capture/v2 version.
But I realized from this comment [here](https://github.com/publiclab/spectral-workbench.js/issues/265#issue-1003375878) that geocoding of the spectrum doesn't work anymore. 
Please I would like to understand exactly what the geocode button is expected to do. Thank you 
[image]

Georjane, 2022-01-18:
I was also facing the error below when I added the geocode button from the old capture interface to the new capture v2. I tried debugging but to no avail. 😞 
[image]

Tlazypanda, 2022-01-18:
Hey @Georjane the Geocode part not working is what we are discussing on the call - the part where I mentioned that the params are not getting saved correctly on to the backend which is why it stopped working. Hence it's not included in the second version so we need to add it and get it working :sparkles: 

This error suggests that data variables passed as an argument here is actually nil, can you log the data variable here reference - https://stackoverflow.com/questions/57197682/in-decode64-undefined-method-unpack1-for-nilnilclass-nomethoderror

jywarren, 2022-01-18:
I believe the geocode button requests a location from the browser API for location, then it tries to attach it to the form using hidden inputs with ID #lat and #lon:

https://github.com/publiclab/spectral-workbench/blob/8f89e0fffd0ac02068857ed6fb775aa5a4460591/app/assets/javascripts/capture.js#L340

Hope that helps! It uses a callback so the code is a little tangled, but it's not too long!

Georjane, 2022-01-19:
> Hey @Georjane the Geocode part not working is what we are discussing on the call - the part where I mentioned that the params are not getting saved correctly on to the backend which is why it stopped working. Hence it's not included in the second version so we need to add it and get it working ✨
> 
> This error suggests that data variables passed as an argument here is actually nil, can you log the data variable here reference - https://stackoverflow.com/questions/57197682/in-decode64-undefined-method-unpack1-for-nilnilclass-nomethoderror

Hi @Tlazypanda 
please i have difficulties understanding this line of code `$('#geotag').val('true'!=$('#geotag').val())`
Thank you!

jywarren, 2022-01-19:
I believe it's ensuring that the value is either true or false, and if it's
not true it automatically defaults to false! A bit convoluted, though!

closed 2022-02-08 by jywarren