publiclab/spectral-workbench #859: Add routes test for spectrums/plots_rss

Tlazypanda, 2022-01-18:
Hi, this is a [first-timers-only issue](https://publiclab.github.io/community-toolbox/#r=all). This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [help wanted](https://github.com/publiclab/plots2/labels/help-wanted) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 📝 **Update** the file [routes_test.rb](https://github.com/publiclab/spectral-workbench/blob/main/test/integration/routes_test.rb) in the `spectral-workbench` repository (press the little pen Icon) and edit the line as shown below. 

[See this page](https://publiclab.github.io/community-toolbox/#r=all) for some help in taking your first steps!

Below is a "diff" showing in read (and a `-`) which lines to remove, and in green (and a `+`) which lines to add:

```diff
+  test "test get request for spectrums plots_rss" do
+    assert_routing({ path: '/spectrums/plots_rss', method: 'get' }, {controller: 'spectrums', action: 'plots_rss'})
+  end
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:

1. If you are familiar with the terminal or would like to learn it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

2. You can also [edit files directly in your browser](https://help.github.com/articles/editing-files-in-your-repository/) and open a pull request from there. 

- [ ] 🏁 **Done** Ask in comments for a review :)

### 🤔❓ Questions?

Leave a comment below!

### Is someone else already working on this?

We encourage you to link to this issue by mentioning the issue # in your pull request, so we can see if someone's already started on it. **If someone seem stuck, offer them some help!** Otherwise, [take a look at some other issues you can help with](https://publiclab.github.io/community-toolbox/#r=all). Thanks!

(This issue was created by [First-Timers-Bot](https://github.com/hoodiehq/first-timers-bot).)

Simba-97, 2022-01-18:
Hey @Tlazypanda can I work on this issue?

Tlazypanda, 2022-01-19:
Hey @Simba-97 please go ahead :)

Simba-97, 2022-01-19:
Sure, thank you for assigning it to me.

Simba-97, 2022-01-28:
@Tlazypanda I believe the task under this is done and the code has successfully been merged, You can close this issue as of now

TildaDares, 2022-02-22:
Closed in #872

closed 2022-02-22 by TildaDares