publiclab/spectral-workbench #869: Add a route helper, profile_path, for user profile

Georjane, 2022-01-19:
Hi, this is a [first-timers-only issue](https://publiclab.github.io/community-toolbox/#r=all). This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [help wanted](https://github.com/publiclab/plots2/labels/help-wanted) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 📝 **Update** the file [routes.rb](https://github.com/publiclab/spectral-workbench/blob/main/config/routes.rb) in the `spectral-workbench` repository (press the little pen Icon) and edit the line as shown below. 

[See this page](https://publiclab.github.io/community-toolbox/#r=all) for some help in taking your first steps!

Below is a "diff" showing in red (and a `-`) which lines to remove, and in green (and a `+`) which lines to add:

```diff
  # Registered user pages:
- get '/profile', to: 'users#show'
- get '/profile/:id', to: 'users#show'
+ get '/profile/:id', to: 'users#show', as: :profile
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:

1. If you are familiar with the terminal or would like to learn it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

2. You can also [edit files directly in your browser](https://help.github.com/articles/editing-files-in-your-repository/) and open a pull request from there. 

- [ ] 🏁 **Done** Ask in comments for a review :)

### 🤔❓ Questions?

Leave a comment below!

### Is someone else already working on this?

We encourage you to link to this issue by mentioning the issue # in your pull request, so we can see if someone's already started on it. **If someone seem stuck, offer them some help!** Otherwise, [take a look at some other issues you can help with](https://publiclab.github.io/community-toolbox/#r=all). Thanks!

(This issue was created by [First-Timers-Bot](https://github.com/hoodiehq/first-timers-bot).)

TildaDares, 2022-01-19:
@Georjane you forgot to add the change for `get '/profile', to: 'users#show'`

Georjane, 2022-01-19:
> @Georjane you forgot to add the change for `get '/profile', to: 'users#show'`

Hi @TildaDares 
I added it and got this error
[image]

TildaDares, 2022-01-19:
@Georjane Is the `/profile` route & `/profile/:id` route used for the same thing?

Also, you can't give two routes the same name. That's what the error is saying

Georjane, 2022-01-19:
Ahh yes, I see. I think the `/profile` and `/profile/:id` are supposed to do the same thing because you can only access a user's profile via their `:id`
That is what I thought initially, reason why I left out `/profile/` route

Georjane, 2022-01-20:
> So I guess you can remove `/profile`.

Okay @TildaDares 
Thank you!
So does this issue count as a valid fto?

TildaDares, 2022-01-20:
@Georjane It does

Xuewei-Chen, 2022-01-20:
Hi, I am taking a course on software development methodology and need to work on some projects. Can I be assigned of this work?

Georjane, 2022-01-21:
@Xuewei-Chen yes you can!

Xuewei-Chen, 2022-01-23:
Hi, I think I have done with this. Can I get a review?

Georjane, 2022-01-27:
> Hi, I think I have done with this. Can I get a review?

Can you please add a link to the PR you opened for this issue here so that you could get a review from @TildaDares @Tlazypanda @jywarren

Midas847, 2022-02-21:
Hey, I am new to this project. Can I work on this issue?

TildaDares, 2022-02-21:
Hi @Midas847, since we haven’t gotten any updates from @Xuewei-Chen, you can take this up. Thanks!

TildaDares, 2022-02-28:
Hi @Midas847, are you still interested in taking this up?

Sandip-Ghosh-2001, 2022-03-08:
I have created a another pr. please check it.

closed 2022-03-10 by Georjane

Sandip-Ghosh-2001, 2022-03-14:
I would love to work. It will be great if you assign me with frontend web. Thanks...