publiclab/spectral-workbench #885: Reproduce detection of fluorescent lights from legacy capture interface into new capture interface (this then offers to calibrate, with an alert along the top!)

Georjane, 2022-02-08:


jywarren, 2022-02-08:
So we found that `is_cfl` maybe is not even used??? Because the line of code actually run is `observe_contrast()` which checks the average amount of "steepness" or contrast in the data every half second. I bet this run faster and more reliably than actually checking if the data matched a fluorescent light pattern, which is interesting:

https://github.com/publiclab/spectral-workbench.js/blob/4e3c9f65d51d3dacd4a831ab2aebf39181620db8/examples/capture/api.js#L41-L64

So, that should be something we can integrate here!

jywarren, 2022-02-08:
And here it is in the new capture code. Is it actually running?

https://github.com/publiclab/spectral-workbench/blob/98014b8db34d933a637ecafeff40712fb750ae3b/app/views/capture/index2.html.erb#L188-L195

closed 2022-02-15 by jywarren