publiclab/spectral-workbench #982: Publishing to stable server

jywarren, 2022-08-02:
Just taking notes on the process; previously it failed as the gems were not installed, we saw the message:

```
[33mweb_1  |[0m Warning: the running version of Bundler (1.17.2) is older than the version that created the lockfile (1.17.3). We suggest you upgrade to the latest version of Bundler by running `gem install bundler`.
[33mweb_1  |[0m The following gems are missing
[33mweb_1  |[0m  * rake (13.0.6)
... many gems
```

https://ci-j.publiclab.org/job/Spectralworkbench-Stable/177/console

Now, I tried to install bundler -- the `gem install bundler` in the Dockerfile had been skipped as it was cached, so i successfully upgraded to Node 16, flushing the cache for the line in question:

https://github.com/publiclab/spectral-workbench/pull/981

https://github.com/publiclab/spectral-workbench/blob/ba11c83aaf8be5cf613056e5a36aec4a92450688/Dockerfile#L43-L46

Now I see this in the [logs](https://ci-j.publiclab.org/job/Spectralworkbench-Stable/178/console):

```
[0mSuccessfully installed bundler-2.3.19

...

[36mweb_1  |[0m Web server started
docker-compose logs
Attaching to spectralwb_stable_web_1, spectralwb_stable_db_1
[33mweb_1  |[0m Warning: the running version of Bundler (1.17.2) is older than the version that created the lockfile (1.17.3). We suggest you upgrade to the latest version of Bundler by running `gem install bundler`.
[33mweb_1  |[0m The following gems are missing
[33mweb_1  |[0m  * rake (13.0.6)
```

So I think we are upgrading bundler in the wrong place; we should do it within the container?

jywarren, 2022-08-02:
Tried it within start.sh here: https://github.com/publiclab/spectral-workbench/commit/2409b1ffa06b957f679fcf230560800ef3806008

jywarren, 2022-08-02:
Oof, so now its really confusing -- 

```
[33mweb_1  |[0m Successfully installed bundler-2.3.19
[33mweb_1  |[0m 1 gem installed
[33mweb_1  |[0m Warning: the running version of Bundler (1.17.2) is older than the version that created the lockfile (1.17.3). We suggest you upgrade to the latest version of Bundler by running `gem install bundler`.
```

jywarren, 2022-08-02:
So apparently last time we just manually changed the version to match what's in the container. I'll try that again.

jywarren, 2022-08-02:
https://github.com/publiclab/spectral-workbench/commit/bc8dc17eef25a197df0b194524428f8467027d95

jywarren, 2022-08-02:
:-/ still getting:

```
[33mweb_1  |[0m Successfully installed bundler-2.3.19
[33mweb_1  |[0m 1 gem installed
[33mweb_1  |[0m The following gems are missing
[33mweb_1  |[0m  * rake (13.0.6)

```

then later can't find rails.

jywarren, 2022-08-02:
Strange - we do this without apparent trouble in MapKnitter and plots2! Both of those are also on up to date `v2.3.16` bundler:

https://github.com/publiclab/plots2/blob/3b7222e5655e41ea4e7102ddbdcb1510932b6228/Gemfile.lock#L810

icarito, 2022-08-05:
I tried adding `bundle update` in start.sh - i know it's not ideal / but it worked.
So first I'm trying a PR with the updated Gemfile.lock, maybe that's all that's needed?

PR #984 (2022-08-05): Try as fix for issue with wrong bundler / not installed gems
Fixes #982 / at least it intends to - this Gemfile is the product of running `bundle update` within start.sh (that got unstable to build and run)

Make sure these boxes are checked before your pull request is ready to be reviewed and merged. Thanks!

* [ ] tests pass -- `rake test`
* [ ] code is in uniquely-named feature branch, and has been rebased on top of latest master (especially if you've been asked to make additional changes)
* [ ] pull request are descriptively named
* [ ] if possible, multiple commits squashed if they're sma [... cut]
closed 2022-08-05 by jywarren